Ultimaker/Cura #12840: Improved Gyroid Infill

limaciform, 2022-07-23:
### Is your feature request related to a problem?

Gyroid infill, in its current implementation, is not as aesthetically pleasing, nor as mechanically optimal as is could be, due to the gaps/holes in the horizontal surfaces of the structure. (Details below)

### Describe the solution you'd like

Keep the current gyroid infill implementation if desired, but add an option or two, for proper gyroid infill that gives the gyroid structure some thickness, so the slicer will fill the horizontal gaps/holes. The new Arachne engine will help greatly with this. (Details below)

### Describe alternatives you've considered

There are no alternatives, as modifying slicer settings to try and fill these gaps/holes merely deforms the gyroid further in the X and Y direction, whilst still not giving any structural thickness to the horizontal sections. All slicers suffer the same problem, but Cura should lead the way for innovation. (Details below)

### Affected users and/or printers

Everyone will benefit from this by having stronger, truly isotropic 3D prints, with the bonus of having aesthetically pleasing gyroid structures to use for all kinds of exposed infill projects. (Details below)

### Additional information & file uploads

### Preface

I’ve been 3D printing for about 7 years, designing countless functional parts for experimentation purposes, personal use, and customer requests. For the majority of those years, I’ve invested my time into mastering PETG as my preferred material of choice, for various reasons which I won’t detail here.

I am always looking for ways to improve performance and print quality, and one of the last details I can improve upon, is material build up on the nozzle. This isn’t a major, print ruining issue; more of a fine detail that will occasionally result in a blob creating minor cosmetic imperfections (I’m very fussy).

I have always used cubic subdivision as my infill of choice, because of it’s ability to build denser cells nearer to the shells and skins (where support is required for a nice surface finish and mechanical strength), whilst building larger cells towards the center of the model, using filament more efficiently.

However, in my efforts to improve the aforementioned minor blobbing issue, I am exploring ways to minimise/eliminate the crossing of extrusion paths, which has led me back to considering gyroid infill as an option.

I experimented with gyroid infill when it was first released, but was left underwhelmed by it’s mechanical performance and print quality, so I left it on the sidelines in the hope that it would be developed and improved into a more useful infill over time.

Unfortunately, nothing much has changed since its release, and from my searches, there hasn’t been a lot of discussion about it either. Hopefully I can change that.

The reason I’ve added this preface, is to give some insight into my experience, and to hopefully avoid sounding like a demanding newbie. Like everyone here, I am constantly pursuing improvements. Unfortunately, I don’t have the coding skills to even attempt to improve the gyroid infill in any slicer program myself.

### The Problem

Gyroid infill is, understandably, a popular choice. It boasts claims of efficient extrusion paths, a good strength to weight ratio, and nearly isotropic mechanical performance. It also looks fascinating.

There are several videos on YouTube that aim to demonstrate it’s advantages, and various 3D printing websites echoing these claims. The problem is, many of the perceived advantages of gyroid infill are just regurgitated from a small number of repeatedly referenced experiments, or taken from other websites and news articles, with titles such as:

“_Researchers design one of the strongest, lightest materials known_” (https://news.mit.edu/2017/3-d-graphene-strongest-lightest-materials-0106)

These other websites and news articles give reference to research papers and scientific experiments. This information is then assumed to apply to the gyroid structures generated for 3D printed infill. But, they fail to discuss the clearly visible differences between the gyroid structures studied in these research papers, and the flawed gyroid structures generated for 3D printed infill.

Below is an image of the gyroid structure studied in the MIT research paper and news article referenced above.

[image]

And here is an image from another research paper titled “_Bio-Inspired 3D Infill Patterns for Additive Manufacturing and Structural Applications_” (https://www.mdpi.com/1996-1944/12/3/499/htm).

[image]

Now if we compare these images to the gyroid structure generated for 3D printed infill below (40x40x40mm cube, 10% infill density, 0.6mm line width, 0.2mm layer height), we see that there are clear gaps/holes in the horizontal surfaces of the structure, that aren’t present in the gyroid structures subjected to scientific experimentation above. These gaps/holes also change shape as the layers progress.

[image]

There has been some debate as to whether or not these gaps/holes matter to the structures mechanical performance. Many people have said it doesn’t matter, but again, only referred back to the same few experiments mentioned earlier.

I think it’s obvious that these gaps _do_ mechanically matter. The gyroid structure works like a 3D honeycomb, absorbing, dividing, and dispersing the stress loads throughout, as seen in the image below from the article titled “_Nature-Inspired, Ultra-Lightweight Structures with Gyroid Cores Produced by Additive Manufacturing and Reinforced by Unidirectional Carbon Fiber Ribs_” (https://www.mdpi.com/1996-1944/12/24/4134/htm).

[image]

If this was a truss style lattice, removing a truss would weaken the structure. But a gyroid is not a truss style lattice; it’s a minimal surface i.e. a 2D sheet formed into the 3D gyroid shape. If we punch holes into that 2D sheet, we’ve created weak points where the sheet can tear, so to speak.

So, the ideal gyroid structure should be free of gaps/holes/defects to achieve maximum performance and true isotropic strength. 

In the image below, we try to close these gaps/holes, achieve a consistent gap/hole size, or obtain more consistent extrusion paths for these horizontal sections throughout the model. Using infill density alone and keeping the same settings mentioned previously, only one density (19%) really comes close. Perhaps there’s a mathematical reason for this, or maybe the consistent pattern breaks down for a taller model with many more layers (I’ve only tested up to the 200mm height of my printers build volume).

[image]

[... 34 lines cut ...]
This image from Print3DD (https://www.print3dd.com/gyroid-infill/) shows the horizontal gaps/holes and poor aesthetics of the current gyroid infill.

[image]

Network (b) – Effectively makes one side of the 2D gyroid sheet into a solid filled lattice structure. I imagine the layer slices would look something like this, but each of the solid section outlines would need to be placed inside the solid half of the 2D gyroid sheet plane, to avoid encroaching in to the hollow sections, and to ensure that the solid infilled area is the equal to the unfilled area. Best for strength. Great for aesthetics. Worst for speed and weight.

[image]

This image from the paper titled “_Additive manufacturing technology for porous metal implant applications and triple minimal surface structures: A review_” (https://www.researchgate.net/publication/330031812_Additive_manufacturing_technology_for_porous_metal_implant_applications_and_triple_minimal_surface_structures_A_review) shows another view of this network solid phase of gyroid, as well as a view of the solid sheet/matrix unit cell discussed next.

[image]

Matrix (c) – Transforms the 2D sheet of the gyroid structure into a 3D sheet with thickness in the X, Y, and Z planes equal to the chosen line width, altered by the infill line multiplier. I imagine the layer slices would look similar to the current gyroid layer slices, with the exception of the horizontal sections, where the lines would transition together at an earlier layer, linger/overlap slightly to achieve the required horizontal thickness, then change direction to transition apart at a later layer. This horizontal thickness would effectively squash the deformed, oval shaped front/back/side horizontal pores downward/upward, to make them circular again. This should be the ideal gyroid structure, and shouldn’t impact speed too much. Great for a balan [... cut ...]
As I mentioned in the preface, I don’t have any coding skills to even know where to begin experimenting with this. All I can presume is that if the current gyroid infill implementation is derived mathematically, then the other gyroid structures can be derived mathematically as well; especially since there are 3D models of the other gyroids in existence already. And all I can hope, is that my explanations are clear enough to understand, to try and give some good enough reasons to push this up the priority list, and appeal to the geniuses here that will know how to execute this.

Thank you for reading, and happy printing!

discip, 2022-07-23:
@limaciform
To be fair: I did not read everything you wrote above, thus this is maybe not exactly what you are aiming for, but what about increasing **`Infill Line Multiplier`**? 😃

GregValiant, 2022-07-23:
That's a heck of a writeup.  I certainly can't complain about "insufficient information" like we often get here.

Infill by it's nature is generally covered on the top, bottom, and all around.  I would think that being "Aesthetically Pleasing" is  very subjective and would be transitory at best.  I have to believe that in virtually 100% of FDM prints the infill rapidly transitions to "Invisible".
I never use Gyroid because it imparts no strength to a model.  I'm with @discip on this one.  Give me Grid at 20% density and with the line multiplier at 3.  That way I can park my car on the print.

limaciform, 2022-07-23:
> 
@discip 
I did cover that lol (just above the zoomed in layer line picture). TLDR; the infill line multiplier works the same as increasing line width, which only effects the X and Y dimensions, not the Z. It just kind of squashes the gyroid in the X and Y directions, deforming the horizontal pores. The lines at the horizontal areas will only ever touch, but there will never be any meaningful thickness/structure/rigidity added, because the gyroids structure as used here, is a 2D sheet trying to be translated in to 3D, so it has no Z thickness in the eyes of the slicer.

It might not even be possible to fill these horizontal gaps with a single line pass, especially as the infill density decreases and the cell size increases. This video here (https://m.youtube.com/watch?v=Djm1sv6rKTQ) sort of shows what I meant above when I said that the gaps/fill areas would need to meet, linger until filled for a few layers, then separate.

limaciform, 2022-07-23:
> That's a heck of a writeup. I certainly can't complain about "insufficient information" like we often get here.
>
> Infill by it's nature is generally covered on the top, bottom, and all around. I [...]

@GregValiant 
Well I'm glad if too much information is the problem lol.

I agree that infill is generally covered on all sides, but... (see below)

Likewise it's generally agreed that, with the exception of long spanning, load bearing surfaces, shell thickness has a greater impact on part strength than infill density. Which is effectively what using grid infill with a 3x line multiplier is doing; treating the infill lines like shells and increasing their thickness.

But... as far as the "aesthetically pleasing" argument goes, I think it's like the chicken and the egg. Is infill covered 100% of the time because that's its nature/purpose? Or is infill covered 100% of the time because its ugly?

I understand that "aesthetically pleasing infill" doesn't sell infill changes as a high priority area of improvement. I'd counter that with two points:

1) Without being able to study my 3D printed parts in a laboratory, under a microscope, with finely tuned scientific equipment, the best indicator of print quality/strength I have, is my eyes. Besides looking for split layers, curled corners, or blobs, I need to see a good surface finish and correct colour to know that I had no temperature/cooling issues throughout the print, so it should perform acceptably and results should be consistent across models. In the same way, I monitor my prints throughout, and should see the same results in my infill, to be sure that properties are consistent throughout each individual model. So to stick with the topic, if I'm chasing isotropic mechanical strength in a part, without crossing extrusion paths, and so choose to use gyroid infill, then when I monitor my print and see sagging extrusions and gaps in the horizontal surfaces, with deformed horizontal pores of what should be an isotropic gyroid, then I know the advertised performance will be compromised.

2) The only part of the whole 3D printing process a customer generally sees, is the end result. Whether its a mesh grill that needs air flow or drainage, or you just want to give something an eye-catching design, exposed infill can be useful in achieving this quickly and easily. People love fancy patterns, hence why honeycomb has always been such a popular design (another infill that the main branch of Cura doesn't have, unlike every other slicer). Let's say you wanted to sell a part you could park your car on, a wheel chock for example, which is just a simple wedge shape. You print a fully covered, 3x line multiplier, grid infill that could hold 5 cars. Your competitor prints one with exposed honeycomb infill of the sides that could only hold 2 cars. Which one will customers likely pick at face value? Probably the honeycomb one because it "looks cool and bees do it and they seen on Facebook that its the strongest structure ever".

So yeah, IMHO, I think infill being aesthetically pleasing does matter.

GregValiant, 2022-07-23:
_"Which one will customers likely pick at face value?"_

The cheapest.

limaciform, 2022-07-24:
> _"Which one will customers likely pick at face value?"_
> 
> The cheapest.

Haha, good joke. If only that were true of people and governments, then perhaps the world wouldn't be so heavily debt laden whilst attempting to cross the frayed rope bridge that is the global economy right now. But, approval ratings and Instagram "likes" (appearances) are what matters nowadays, no matter the cost.

If you want to use the cheapest cost argument though, with all settings the same other than the infill pattern and 0 top/bottom layers, then it looks like exposed honeycomb infill wins again.

[image]

Joeydelarago, 2022-07-25:
I might frame this feature request and put it up on my wall!

limaciform, 2022-07-25:
> I might frame this feature request and put it up on my wall. Thank you for the thorough explanation!

Haha feel free. Just be careful it doesn't send you crazy! Then it will keep you up at night, like it does to me lol.

Thanks for taking the time to read through and consider my points!

limaciform, 2022-07-26:
Just wanted to add a few more bits of information to this.

The image below, taken from this paper (https://www.researchgate.net/publication/358758848_Numerical_Modeling_and_Experimental_Investigation_of_Effective_Elastic_Properties_of_the_3D_Printed_Gyroid_Infill), shows the stresses in a solid lattice network type gyroid when applied with a longitudinal and shear strain. The key point from this one is where the strain is placed, i.e. exactly where the gaps/holes in the gyroid infill are placed.

[image]

In my initial post, I spoke about the issue with the horizontal gyroid surfaces having zero thickness, and the possibility of using line width as the measure for horizontal thickness. I also spoke about having 3 gyroid type options. But, the images below have given me a more refined idea. Taken from (https://www.researchgate.net/publication/311583144_Mechanical_properties_of_a_new_type_of_periodic_interpenetrating_phase_composite_materials).

[image]

When gyroid infill is selected, three options could appear: 

- "Sheet Thickness" (mm) (set to 0mm by default)
- "Solid Volume" (%) (set to 0% by default, overrides "Sheet Thickness" if >0%)
- "Invert Volume" (tick box)
 
"Sheet Thickness" with a setting of 0mm, would be exactly the same as the gyroid infill behaves now. A setting equal to line width mm, would work the same as gyroid infill does now, but there would be horizontal thickness added, so horizontal gaps/holes would be filled, and horizontal pores would be circular again. A setting greater than line width, will make the sheet even thicker, and the extrusion lines would merge/split as needed per Arachne's smart slicing (the "Infill Line Multiplier" option would disappear, as it would not be necessary). Refer to the top half of the above image, regarding "sheet-networks" and sheet thickness.

"Solid Volume" with a setting >0% would override the "Sheet Thickness" setting (disable, grey out), and would transform the gyroid into the solid lattice network type. This could be increased to anything <100% (solid) for a thicker solid lattice network. Refer to the bottom half of the above image, regarding "solid-networks" and solid volume.

"Invert Volume" would be a simple tick box. This would swap which half of the solid lattice network type gyroid, contained the solid lattice. Refer to "Volume 1" and "Volume 2" in the image below.

[image]

The "Sheet Thickness" solution is the most important feature for fixing the gaps/holes and deformation issue. It can be applied to any other triply-periodic minimal surface types of infills added in future as well. As tested in Smartavionics Cura build, Schwarz-P and Schwarz-D infills also suffer from the same issues.

The solid lattice network type of gyroid would be nice to have, but it all depends on how difficult it is to actually implement it, and obviously, priorities and resources.

smartavionics, 2022-07-29:
With regard to filling in the little holes in the peaks of the gyroid infill, I've been doing some experiments. I can now reliably get it to produce infill that has a cross in the peaks rather than a hole like this slice shows..

[image]

Here's a few more layers...

[image]

To achieve this I found I needed to constrain the pitch of the gyroid to be a multiple of 4 times the layer height. A side effect of applying this constraint is that it becomes impossible to obtain all densities. I think this is due to the fact that the z coordinate changes in discrete (layer height) steps.

Whether this is actually worth using, I don't know but I'll probably make it an option in my next Cura build.

I won't be implementing the solid lattice gyroid because it's way beyond my mathematical capabilities to work out how to generate it.

limaciform, 2022-07-29:
First of all, thank you for taking the time to read through and experiment with this! It's greatly appreciated!

> With regard to filling in the little holes in the peaks of the gyroid infill, I've been doing some experiments. I can now reliably get it to produce infill that has a cross in the peaks rather than a hole like this slice shows..
> 
> [image]
> 
> Here's a few more layers...
> 
> [image]
> 

How do the extrusion paths look? Does it extrude lines on one diagonal, then the other, bridging the gap, but crossing paths? Or is it still running X/Y and changing direction at the peaks?

It's a bit of a trade off either way as to whether paths cross and potentially add nozzle build up, or direction changes causing extrusions at the peaks to sag; paths are probably going to cross to get a (close to) perfect gyroid anyway. It looks like an improvement. My only nitpick with this solution is that the peaks are still only the thickness of the layer height, not the extrusion width, so the sheet thickness for the gyroid still isn't (close enough to) uniform.

> To achieve this I found I needed to constrain the pitch of the gyroid to be a multiple of 4 times the layer height. A side effect of applying this constraint is that it becomes impossible to obtain [...]

That's well beyond my mathematical abilities lol. But the Z coordinate changes not aligning exactly with the gyroid formula/structure/peaks at particular layer heights with every density does make sense to explain why the shape of the gaps changes currently.

> Whether this is actually worth using, I don't know but I'll probably make it an option in my next Cura build.
> 

It'll be great a great option to experiment with. That's half the fun of 3D printing!

> I won't be implementing the solid lattice gyroid because it's way beyond my mathematical capabilities to work out how to generate it.

That's fine. It was just another idea. To be honest, I think the sheet network type would probably still get the most use anyway.

I'm at work at the moment. But when I get home tonight, I was planning to experiment a bit and see how Cura/Arachne goes slicing a gyroid with a sheet thickness the same as line width. I'm curious to see how it treats the gaps/peaks.

smartavionics, 2022-07-30:
> How do the extrusion paths look? Does it extrude lines on one diagonal, then the other, bridging the gap, but crossing paths? Or is it still running X/Y and changing direction at the peaks?

That's a very good question. The existing code for determining the order in which the infill line segments are printed doesn't take into account any change in nozzle direction when moving from one line segment to another. Therefore, I have introduced a penalty factor based on the amount the direction of the next candidate line differs from the last line printed. So it will now favour the wavy lines on the diagonal and will try to avoid the right angle changes of direction at the peaks.

Thanks for the idea!

limaciform, 2022-08-04:
> That's a very good question. The existing code for determining the order in which the infill line segments are printed doesn't take into account any change in nozzle direction when moving from one [...]

Awesome! Not a problem. Happy to help with anything that gets us all better (or just cool) results.

Sorry I haven't been back for a while. I've been thinking hard about this. I'll dump a bunch more thoughts in the next post.

limaciform, 2022-08-04:
### Letting the Slicer do the Work

I downloaded a 20mm gyroid cell with a sheet thickness of 0.5mm, to see how it slices. I used a layer height of 0.2mm, and line width of 0.6mm in the images below, but also tried different line widths without achieving any better results.

[image]

It doesn’t look too bad at first, but there are a few issues. From left to right; the heavily sloping sections slice thicker so introduces sections with two extrusions, some extrusions want to start on unsupported areas (might be a broader improvement opportunity here for the slicer in general), and we still get those sharp direction changes on unsupported areas (perhaps another improvement opportunity in general).

[image]

Shrinking the model gives almost the same kind of results. Scaling the model up makes the peaks/overhangs even worse (image below), but seems to solve the other two issues.

[image]

Based on all the experimentation and information up to this point, and given the way the slicer behaves as of writing this, the conclusion is that fixing that gaps/holes completely, is not necessarily as simple as giving the gyroid infill some sheet thickness and letting the slicer do the work.

### Researching the Math

I decided to keep researching to try and get a better understanding of the math behind these gyroids. I am by no means claiming to be an expert all of a sudden; I’m sure there’s already some mathematicians here that don’t need this. I’m also not trying to explain it like you (the reader) are five years old. But I’ll share what I think I understand so far, in case it’s of any help to anyone, and so I can think out loud for my own benefit.

**Gyroid Unit Cell**

The first thing I think it’s important to understand, is what a gyroid “cell” is. To use an image referenced previously, these are gyroid “cells”.

[image]

We can imagine them as cubes. If we were to individually slice these cubes/cells in to 2D sheets, almost every slice would look different. The only slices that would be the same, are the outermost opposing faces. This allows these cells to stack together to form a matrix/array/network (whatever term you would prefer to describe it as). So when we refer to “unit cell length”, we are describing the X, Y, and Z dimensions of the cell/cube. The other way to think of it, is as pattern frequency i.e. how far you would have to travel along the X, Y or Z axis before the pattern starts to repeat itself. So in the images below, we have a unit cell length of 3.

[image]

**Iso-value**

The next part, is what’s referred to as the “iso-value”. The best way I can describe it for now, is as a sort of offset/shift value. So if we slice up a gyroid and look at one of the layers, we see a group of identical wavy lines (see images below). If we have an “iso-value” of 0 (green), the wavy lines remain unchanged (0 is the default iso-value for a minimal surface gyroid). If we change the iso-value (+/- 0.5), the wavy lines shift to either side of their original position, and slightly change shape with a change in ‘z’, depending on if a positive (blue) or negative (red) iso-value is used.

[image]

Now we have a rough idea of those two pieces of the puzzle, we can take a look at the functions.

### The Functions

This paper (.pdf download) titled “Increased efficiency Gyroid structures by tailored material distribution” (https://www.researchgate.net/publication/343887451_Increased_efficiency_Gyroid_structures_by_tailored_material_distribution#read) contains a lot of information, such as this interesting image showing the internal helical structure that gives the gyroid it’s almost isotropic, spring-like resistance

[image]

Most importantly, it contains all of our functions, accompanied by a helpful description of how surface (i.e. minimal surface), sheet solid (i.e. solid-surface form), and solid network (i.e. solid-network form) gyroids are formed.

[image]

From what I can gather:

_Eq. 2_ = a surface (minimal surface) gyroid, as is currently used for infill.

_Eq. 3_ = a solid network (solid-network form) gyroid, where the region ‘<h’ (or ‘>h’ to invert) is selected/filled (i.e. the region to one side of the gyroid lines graphed in Eq. 2).

_Eq. 4 and Eq. 5_ = a sheet solid (solid-surface form) gyroid, where the regions either side of the gyroid lines graphed in Eq. 2, are selected, with the gyroid lines in Eq. 2 being the center lines, and the iso-value ‘h²’ varying the sheet thickness.

Translating these functions to relate to 3D printing; ‘x’ and ‘y’ are simply our X and Y coordinates, ‘z’ is out Z coordinate or cumulative print height, ‘h’ in it’s variations (i.e. =, <, >, ²) would be our control for the gyroid type/form, sheet thickness, or inverting the solid portion of the solid network gyroid, and ‘a’ would be our control for unit cell length as a function of infill density (I will come back to this later in this post).

### Eq. 2 – Minimal Surface Form Gyroid

If we graph Eq. 2, we see the direction change in the lines where the peaks of the gyroid would be printed, which just reiterates those variations that give us gaps in these surfaces as the print height progresses.

**Function:**
sin((2π)/a*x)*cos((2π)/a*y) + sin((2π)/a*y)*cos((2π)/a*z) + sin((2π)/a*z)*cos((2π)/a*x) = h

**Where:**
h = 0
a = 3
z = 0, 0.4, and 14  (representing some 0.2mm layer steps)

[... 334 lines cut ...]
With a sheet solid form gyroid (not selecting the sheet region, just the outlines):

**Function:**
[... cut ...] izontal thickness, avoid overly sharp, unsupported direction changes, and keep extrusion paths relatively tidy.

I hope all of that makes some sense. I’ve written numerous things, and held backspace to delete it, over and over again. And I could hold off posting until I think some more and have some definite conclusion, but knowing me, that could takes months. So, I think it’s better to just vomit it all out so far, and see if this sparks some ideas, or if it could lead to some useful discovery.

limaciform, 2022-08-05:
I graphed and coloured a sheet solid form gyroid with thickness of approx. 0.6 (to match extrusion width, with a h value calculated from polynomial function in last post), a z step (layer heights) of 0.2, and unit cell length of 8 (to work with the previously discussed pattern features/horizontal surface occurences).

We get 3 bridging layers (5, 6, and 7) to give us our 0.6 horizontal thickness to match our wall thickness. And I think intervening to control the pattern extrusion direction will allow the gaps to be bridged (see rough arrows).

[image]

smartavionics, 2022-08-05:
I've just released https://github.com/smartavionics/Cura/releases/tag/4.20.3, it has a new setting Minimise Gyroid Holes that constrains the pitch so that you always get 2 "crosses" per gyroid cycle. Further more, the crosses are printed as straight lines rather than as right angles so they should print nicely.

limaciform, 2022-08-05:
> I've just released https://github.com/smartavionics/Cura/releases/tag/4.20.3, it has a new setting Minimise Gyroid Holes that constrains the pitch so that you always get 2 "crosses" per gyroid [...]

Thank you! I'm away from home for a week and a bit, but I'll definitely have a play with it when I get back!

Did any of those formulas above look like anything of use to you?

smartavionics, 2022-08-05:
> Did any of those formulas above look like anything of use to you?

Sorry, I'm not looking at implementing multi-layer gyroid infill so I'm letting all this math wash over me!

Cheers,

Mark

limaciform, 2022-08-05:
> > Did any of those formulas above look like anything of use to you?
> 
> Sorry, I'm not looking at implementing multi-layer gyroid infill so I'm letting all this math wash over me!
> 
> Cheers,
> 
> Mark

Alright, no worries.