Ultimaker/Cura #12843: Different seam positions when slicing with same parameters

olive007, 2022-07-23:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Snapmaker A150

### Reproduction steps

1. Add a STL to the project.
2. Duplicate it
3. Set Special `Modes > Print sequence` to "One at a time"
4. Set `Walls > Z Seam Alignment` to `User Specified`
5. Set `Walls > Z Seam X` to `500`
6. Set `Walls > Z Seam Y` to `500`

### Actual results

[image]
We can clearly see the seam randomly dis-align themself for no reason.
[image]
The bug isn't consistent with duplicated model. The unalignment are at different place.

### Expected results

All the seams should be align to one corner. But that is not the case.  
Plus the 2 models have different seams position while they are exactly the same.

The slicer should **ALWAYS** get the same result with same parameters.
- Set the `seam corner preference` to any value and slice.
- Admire the seam placement
- Change the `seam corner preference` to another value and slice.
- Admire the seam placement
- Change back the `seam corner preference` to the first value and slice.
- Admire a different seam placement that the original. (That doesn't occur all the time, repeat in that case)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

While we are at this problem. I'll be nice to have:
 - a visualization of the `User Specified` seams position on the layer view.
 - an option to place the seams inside the object. (to have no seam visible on exterior faces).
 - add multiple  `User specified` seam positions relative to the model. (Cura will chose the closest)
[cura.log](https://github.com/Ultimaker/Cura/files/9174222/cura.log)
[S2A_desk foot top.zip](https://github.com/Ultimaker/Cura/files/9174224/S2A_desk.foot.top.zip)

GregValiant, 2022-07-23:
Thanks for the report.
If you hover the mouse over the setting "Z seam relative" you will see that when it is enabled "...the Z seam coordinates are relative to each parts center."  When Z seam Relative is disabled then the Z seam location is "absolute".  In your project file Z Seam Relative is disabled.

Try setting the "Z Seam Alignment" to "Back Right" (the location of your 500,500) and click on the Function icon next to the X and Y setting to get them back to normal settings values.  Then enable Z seam Relative. and see how that works for you.

olive007, 2022-08-07:
This issue should not be auto close...
No fix have been proposed.
What is the status of the investigation.

GregValiant, 2022-08-07:
Your post has re-opened the bug report.
The status was that I provided options for adjusting the seam and wanted to know if you made those adjustments, and whether or not they worked for you.  That is what the "Status: Needs Info" was for.
As you can see it has been re-opened.  May I assume that you did not find a satisfactory combination of settings?

olive007, 2022-08-14:
I just change the settings like you explained @GregValiant, and not all the seams are aligned.
Look at the last inner wall the seams are on the left while all the others are on the right...
[image]
[image]

GregValiant, 2026-08-19:
SEams have been reworked recently and now (Cura 5.13.0) include Paint On seams.  I'll go ahead and close this.

closed 2026-08-19 by GregValiant