Ultimaker/Cura #12845: Having cura on a monitor left of the primary display causes a hang

g3gg0, 2022-07-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

none

### Reproduction steps

1. have two monitors
2. assign the right one as primary monitor
3. you will find a "window_left = -1614" and "window_state = 2" in your cura.cfg
4. next startup causes an empty window

### Actual results

cura hangs and does not respond anymore

### Expected results

expected *not* to hang on startup

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9174840/cura.log)

clarkiej, 2022-07-24:
[cura.log](https://github.com/Ultimaker/Cura/files/9176593/cura.log)

ASame Issue. Splash screen opens on secondary Monitor, Cura opens on primary monitor. Any secondary window such as marketplace or preferences opens on secondary window and Cura hangs.

SwappingPrimary and Secondary monitors has teh same issue, with software opening position 

Enabling or disabeling "remember Window Position" has no effect

Only way to get Cura to work is to disconnect one screen - Not a practical or effective solution

g3gg0, 2022-07-24:
just edit the cura.cfg and delete the "window_left = xxx" line

clarkiej, 2022-07-25:
> just edit the cura.cfg and delete the "window_left = xxx" line

There is no "window_left" line in my configuration file. There is "settings_xpos = -725.0
settings_ypos = 60.0" and removing them makes no difference on teh crashing of Cura

g3gg0, 2022-07-25:
[image]

This is from my config file

clarkiej, 2022-07-25:
If you enable  "remember Window Position" under General Settings, then it will add this line.
LIke I said, having it disabled, the splash screen still opens on the secondary monitor and Cura opens on teh primary. Secondary windows such as marekety place open on teh secondary window and hang crashing Cura.

The only way to stop this is if I disconnect my second screen Start Cura, then redconnecty my second screen - not teh most practical solution.

g3gg0, 2022-07-25:
if you add this line and set it to zero, will it use that position from the config?

clarkiej, 2022-07-26:
I added this line under [General]

restore_window_geometry = False
window_left = 0
version = 7

And it makes no difference.Teh splace screen still  opens on my secondary monitor, Cura runs on my Primary monitor and opening MarketPlace still causes it to open on my secondary monitor , which hangs and then freezes Cura

MariMakes, 2023-03-31:
Quick update on our side,

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download link and discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)