Ultimaker/Cura #12849: extension for inventor

DavideMarchi92, 2022-07-24:
### Is your feature request related to a problem?

hello, on version 5.1.0 of Cura there is no extension for .ipt files created by inventor.
would it be possible to have it?

### Describe the solution you'd like

insert as in previous versions the possibility to directly load the inventor .ipt

### Describe alternatives you've considered

nothing

### Affected users and/or printers

me

MariMakes, 2022-11-01:
Hey @DavideMarchi92,

Sorry, it took us a while to get back to you 😞 

If you have a Professional or Excellence Ultimaker subscription you have access to the Native CAD import plugin.
There the .IPT file format is supported. You can read more about it here:
https://support.ultimaker.com/hc/en-us/articles/360022116740-Opening-native-CAD-files-in-Ultimaker-Cura

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand. 🙏

closed 2022-11-01 by MariMakes

DavideMarchi92, 2022-11-03:
Hi, ok thank you very much