Ultimaker/Cura #12850: Fuzzy skin not working with spiralize outer contour mode

marleo, 2022-07-24:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Activate fuzzy skin mode as well as spiralize outer contour.

### Actual results

Fuzzy skin only beeing printed on the layers that have a wallthickness bigger than 1.

### Expected results

Both working together, like they did in Cura 4.13.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I printed several parts with those two modes together in Cura 4.13.1 without a problem. (using thickness .1mm and density 5 1/mm)
Cura 5.1.0 doesn't seem to allow it anymore. In the preview window only the layers with a wallstrength bigger than 1 allow fuzzy skin. 
[image]
[image]

Vandresc, 2022-07-26:
Hi @marleo thank you for your report, we've created a ticket on our backlog. 
Devs see Cura-9518 for our internal reference.

kthornbloom, 2022-11-24:
@Vandresc - any updates? Both great features that were fun to use together!

Vandresc, 2022-11-24:
Hi @kthornbloom , currently still on our backlog. Will check the priority with the rest of the team.

Jerooney, 2022-12-01:
Hi @Vandresc , this issue has been reported as a support case. +1 !

Thorvarium, 2023-01-10:
Any updates on this?

Vandresc, 2023-01-12:
Hello everyone, sadly there has been no update and the ticket is still in our backlog. Although this has a top priority as well, there are other critical issues that we want to address before the next release. We cannot guarantee that this will be included in the next release, but if time allows, we will try to include it, and PRs are always welcome, of course.

Goury, 2023-01-13:
F
Also broken in 5.0

wondah85, 2023-04-18:
Still broken in 5.2.1 and 5.3.0 Alpha, used this function so much and following all the messages in the facebook post, others did as well. 

Is this still "top priority"?

C4pnK1rk, 2023-05-04:
Still broken in 5.3.1. I don't understand how this hasn't been fixed. It's a pretty important feature. It seems to be somehow tied to "Bottom Layers" / "Initial Bottom Layers" It will only do "fuzzy skin" for the amount set, but it also ignores "Spiralize Outer Contour".

kthornbloom, 2023-08-09:
@Vandresc Just checking in to see if this can added to the work queue sometime soon? I <3 Cura, so it's a bummer to have to direct people to other slicers for certain models due to this bug.

dgponcegmail, 2024-05-21:
Any comment about this issue ? Thanks in advance. Really used combined featured.

akajester, 2024-07-14:
This explains why I can't get fuzzy skin with spiralize outer to work. Anyone looking at this? Thanks.

kvanonavk, 2024-08-12:
Hello. Any updates regarding this?

solomonBoltin, 2024-09-18:
Is somone working on this ?

Goury, 2024-09-18:
@solomonBoltin no, as the status says

kthornbloom, 2024-10-05:
Went to use these features today. Found my own comment here from _2 years ago_. 🫠 

@nallath Sorry to ping you, just looking for someone active in this repo. Is there a chance this one could get un-backlogg'd?

TQM21, 2025-07-28:
Tried this combo, still noworkee. Using Cura v5.7.2 in July 2025. Three years later, still "On Backlog."  Please consider fixing, thanks.