Ultimaker/Cura #12851: 3D Jake PLA Menu

chasrobertson, 2022-07-24:
### Application Version

5.1 & 5.0

### Platform

Windows 11

### Printer

Weedo X40

### Reproduction steps

When you try to choose the £DJAke PLA material, the drop down list appears, but you are unable to scroll down to the missing itmes at the bottom.
This only happened after move to 5.0 and also is in 5.1

### Actual results

Have tried everything to get at the missing itmes, but have not found a way around this problem

### Expected results

This worked fine on the earlier 4 versions

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No File uploads are necessary inb this instance

GregValiant, 2022-07-25:
Thanks for the report.
I'm not sure how you are accessing the materials.  Could you add a screenshot of the problem?

chasrobertson, 2022-07-25:
Hi,
Here is a screenshot of the 3DJAke Material menu. There are a a great
deal more filaments in this list, but I am unable to scroll through the
items which are off the list at the bottom. I used to be able to do this
on all of the version 4 Ultimakers, but since version 5 came out,
options like white Eco PLA are no longer selectable.

If you need any more information, please get back to me. Thanks for the
prompt reply.

Regards

Chas.Robertson.

>>> GregValiant ***@***.***> 25/07/2022 01:18 >>>
Thanks for the report.
I'm not sure how you are accessing the materials. Could you add a
screenshot of the problem?
—
(
https://github.com/Ultimaker/Cura/issues/12851#issuecomment-1193427868)
(

chasrobertson, 2022-07-25:
Here is the Screenshote you requested

jellespijker, 2022-07-29:
@chasrobertson Is there a reason why you closed this?  Is this no longer an issue?

I will bring this ticket up with the team to be discussed

chasrobertson, 2022-07-29:
Hi, did not realise I had closed this by mistake. Should have been left open  Sorry for the mistake..CheersChas.Robertson Sent from my Galaxy<div>
</div><div>
</div><div>-------- Original message --------</div><div>From: Jelle Spijker ***@***.***> </div><div>Date: 29/07/2022  07:17  (GMT+00:00) </div><div>To: Ultimaker/Cura ***@***.***> </div><div>Cc: Chas Robertson ***@***.***>, Chas Robertson ***@***.***>, Mention ***@***.***> </div><div>Subject: Re: [Ultimaker/Cura] 3D Jake PLA Menu (Issue #12851) </div><div>
</div>

>>> "Jelle Spijker" ***@***.***> 07/29/2022 07:19 >>>
Reopened #12851.

x40-Community, 2022-08-16:
I can confirm the bug. A material selection is not possible if many materials are listed. With Cura 4.X everything still worked.
I ask for a solution to the problem.