Ultimaker/Cura #12854: Hole perimeter printed before the entire surface on top layer

ntfshard, 2022-07-24:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Prusa i3

### Reproduction steps

Load hollow box model with a hole in top surface. And slice it without supports

### Actual results

Slicer prints hole perimeter before the initial layer of surface

### Expected results

Expected not printing in the middle of nothing, surface should be applied first

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[PI3_powerbox-Body.zip](https://github.com/Ultimaker/Cura/files/9175995/PI3_powerbox-Body.zip)
[powerbox-Body.stl.zip](https://github.com/Ultimaker/Cura/files/9175996/powerbox-Body.stl.zip)
[PI3_powerbox-Body.gcode.zip](https://github.com/Ultimaker/Cura/files/9175998/PI3_powerbox-Body.gcode.zip)

smartavionics, 2022-07-24:
My cura variant (see https://github.com/smartavionics/Cura/releases) provides a setting, Remove Holes Above Air. Enabling that will remove the hole in the first layer of the roof of your model as shown below. Obviously, you then have to trim away the filament that covers the hole.
[image]

I have also enabled the experimental bridging settings as they help with the bridging used for the first few roof layers.

ntfshard, 2022-07-24:
@smartavionics, looks very interesting, thanks! But I guess don't need more that 1 of this boxes for my purposes. Due walls is quite thick, the hole is more or less formed, may be I enforce it later (it still printing).

I just want to report a bug =)

GregValiant, 2022-07-24:
Thanks for the report.  Someone from the Cura Team will also take a look.

For myself, I don't see a bug here.  The entire first layer of the top of the box is going down over air.
Changing the print order from "Walls then Skin" to "Skin then Walls" would result in the skin not having anything to attach to on the periphery because the outside of the box would be 1 layer short, and there would be nothing to attach to at the hole because the inner walls aren't there yet.  You would have spaghetti.
Changing the print order to Periphery Walls - Skin - Hole Walls would require a discrimination that doesn't exist in Cura.  The outside of the box is a polygon with 4 sides and the hole feature is a polygon with a lot of sides.  Even if it could be done by adding a setting of some sort, the skin extrusions would end at the hole and would be cantilevered from the peripheral walls.  The ends would fall and you would have a slightly different mess of spaghetti...(linguini?).

@smartavionics fix is a decent one.  You might be able to emulate it by changing the hole so it has a depth of 2 or 3 layers.  That would allow the first and second top skins to go down from wall to wall and then the hole would be present on the remaining layers.  You would still need to trim out the hole.
FDM isn't good at printing over air.  If it was me, I'd design a top for the box and super-glue it on.

I thought about changing this from a "Bug Report" to an "Improvement" or "Feature Request" but I'll leave it as a Bug Report for now.
@ntfshard  Do you have an idea of a change you would like to see made in Cura that could resolve this?

ntfshard, 2022-07-24:
> For myself, I don't see a bug here. The entire first layer of the top of the box is going down over air. (...) FDM isn't good at printing over air. If it was me, I'd design a top for the box and super-glue it on.

Yes, but it should not be a problem in general: the first layer will be stringy and weak, but the next layers will add extra strength.   The top of this model is 2 mm thick, and it's rigid enough (printer completed a job). For more accurate/precise and beautiful tasks glue is acceptable solution, or enabling supports and wasting literally tons of material: 26g just model vs 44g-normal or 37-tree supports.

> Do you have an idea of a change you would like to see made in Cura that could resolve this?

I guess it should be something like: _do not start line/extrusion from the point without any other geometry around. Better to drop this. In other words: we don't care about existence of end point, but have to check the starting point._

For my case, it could be starting all lines from outer shell until the hole. And the first rings of hole should be discarded. (This is the second version of model, in original "bridge" and "tooth" which under the top layer are absent)

[image]

PS
I changed a model a little bit and other problem appear: 
[image]
Starting a line without connectivity to any surface (image above with full shape of this "tooth")

GregValiant, 2022-07-25:
If you discard the walls of the hole you get this situation where all the skin lines end at nothing.
[image]

If you use a cylindrical support and turn off the rest of the supports, it has a chance to finish.
[image]

Regarding the other thing, try changing the "Split Middle Line Threshold" and/or "Add Middle Line Threshold"  Start at 50% for each and work down.  But it's still going to fail.  You can't print over air.

ntfshard, 2022-07-25:
> You can't print over air.

But slicer is trying, that's the initial topic. It prints first 3 circles literally over air. And as I said before, much better if it was just discarded.

> If you discard the walls of the hole you get this situation where all the skin lines end at nothing.

Only the first wall of the hole. Because case of printing over air can be detected by slicing engine, I suppose(at least it has information from previous layer(s) and could try use it). Next layers will see the skin from previous run and keep geometry in place.

Version of model from the first message (with those settings) is printable and rigid enough to work, but messy around the hole (and has wrong dimensions which is not related to this topic).

Not sure how to enable such type of supports. I guess it could be best in cost-quality ratio

GregValiant, 2022-07-25:
That Cylindrical Support shape is from a plugin available in the MarketPlace called "Cylindric Custom Support" by @5axes .  (It provides a few different options for shapes.)  In this case it is set to "Print as Support".  I figured since you have that square hole in one wall that it would be possible to worry out a single support structure with needle nose pliers.  Setting the "Support Bottom Distance" to 2X layer height pretty much insures that the bottom will break free without issue.

Vandresc, 2022-07-29:
Hi @ntfshard, we agree with @GregValiant and there is not a quick fix for this.  Our community is a big part of that by requesting new features and reporting bugs.
To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to defer this bug as we believe we will not work on this anytime soon.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
Thank you for bringing this issue up to our attention.

ntfshard, 2022-07-29:
@Vandresc, Ok, no problem. I only reported a strange behaviour of the tool)