Ultimaker/Cura #12855: Startup Locking program 5.0 and 5.1 - CURA.cfg

EliteGeeksUSA, 2022-07-24:
### Application Version

5.0 and 5.1 both

### Platform

Windows 10

### Printer

Tevo Tarantula i3

### Reproduction steps

I installed, used it and no problem... 
closed program and will run through the steps to restart, then it locks up the program just as the program opens after the loading screen. 
I then go into the cura.cfg file in the folder C:\Users\Owner\AppData\Roaming\cura\5.1  and remove the lines in the General area  " window_top = 203 window_left = 210 "   and then try to restart, and it works again, till I shut it down and restart, and it does it all over again.   
I run 3 monitors.  All drivers are current. It is an 11th Gen i9 with GTX 2080 Super and 64 Gigs of memory with 4 M.2 Drives. - Clean installation of the OS a few months ago. Win 10 64 bit

Problem was in 5.0 and 5.1

### Actual results

Program Locks Up on start

### Expected results

Program to open.  :D 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9176146/cura.log)
[cura.cfg.log](https://github.com/Ultimaker/Cura/files/9176147/cura.cfg.log)

PockyBum522, 2023-09-25:
I also have this issue. It only occurs when I was using Cura on my left monitor and exit the program.

If I change window_left to be a positive value, Cura will open fine on my main monitor. If it is any negative value, program crashes on open with a white screen forever. 

Changing that value from negative to positive is all I ever need to do to fix it. At this point I'd be fine if Cura just checks if it's negative and makes it positive before launching.

Beast76, 2024-09-16:
> I also have this issue. It only occurs when I was using Cura on my left monitor and exit the program.
>
> If I change window_left to be a positive value, Cura will open fine on my main monitor. If it [...]

I had the same problem and tried to find out at what value this happens by trial and error. As soon as the value is **less than -640**, the program no longer opens.

`window_left = -640` works
`window_left = -641` crash