Ultimaker/Cura #12856: Cant Slice after adding kobra max profiles from git.

Jahigs, 2022-07-24:
### Application Version

5.1.0

### Platform

linuxmint 20.3 uma

### Printer

Anycubic Kobra Max

### Reproduction steps

Downloaded the last committed PR Cura-0d4c77d5c0bb8371c7735b863c1fbf9eb2f78663
Grabbed The Following folders from it: Definitions, Extruders, Quality.
Reloaded cura 5.1, added an Anycubic Kobra Max to my printers
Selected Kobra printer, selected suggested parameters profile in print settings
attempted to slice benchy

### Actual results

Slicing Failed with an unexpected error Please Consider reporting.

### Expected results

generate slice for model

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9176332/cura.log)

GregValiant, 2022-07-24:
Thanks for the report.
I'll remove the bug report label because it seems you have used files written for the AnyCubic fork of Cura.  Their fork isn't the same as Ultimaker Cura and likely lags behind making some of their files incompatible.

When adding the files to Ultimaker Cura did you happen to over-write the UM Cura "fdmprinter.def.json" file with one from the AnyCubic site?  The AnyCubic file doesn't contain some settings that are present in UM Cura 5 versions (as noted in the log you provided).  In particular is "wall_transition_filter_deviation" which appears to have caused the slice to fail.

I'll leave this open so you can comment.

Jahigs, 2022-07-26:
You guys rock :) did a fresh install and only added the kobra max's json files instead of dumping the entire folder contents for the 3 folders mentioned above. All is well!

GregValiant, 2022-07-26:
Perfect.  I'll close this then.

closed 2022-07-26 by GregValiant