Ultimaker/Cura #12857: 'Processing Layers 100%' window cannot be dismissed, blocks Preview status

robughblah, 2022-07-24:
### Application Version

5.1.0

### Platform

Mac Big Sur

### Printer

FLSUN QQ-S 

### Reproduction steps

A small box appears near bottom middle od screen when Processing Layers is working to completion. Then the box cannot be dismissed or moved. A second box with same status will appear above it, same persistence
[Screen Shot 2022-07-24 at 18.03.29.pdf](https://github.com/Ultimaker/Cura/files/9176802/Screen.Shot.2022-07-24.at.18.03.29.pdf)
.

### Actual results

cannot see Preview progress bar.

### Expected results

PRocessing Layer status boxes can be dismissed
[cura copy.log.zip](https://github.com/Ultimaker/Cura/files/9176806/cura.copy.log.zip)
.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

ok

marios-stam, 2024-06-17:
Hello, I am having the same problem. Is there any fix to it?

marios-stam, 2024-06-25:
Turns out there was a numpy issue, I downgraded it to `1.23.5 `and now it works fine.