Ultimaker/Cura #12860: Dialog box overwriting a gcode file is not correctly localized (german).

rih2555, 2022-07-25:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3 pro

### Reproduction steps

When slicing and saving the same gcode file a second time to harddisk then appears a dialog box with the question "The file already exists. Do you really want to overwrite it?" My installation is in german. Therefor this dialog is in german, but the buttons are english yes / no, instead of ja / nein.

[image]

### Actual results

The buttons are working as expected. It is only a localization problem. This problem is since I use Cura 4.x and still in Cura 5.

### Expected results

Not relevant. It is just a german localization problem. Please fix it. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No more additional informations.

fieldOfView, 2022-07-25:
AFAIK, the "Yes" and "No" texts here are provided by the operating system, regardless of the language set in Cura.
Could it be that you are running on an english version of Windows?

rih2555, 2022-07-25:
No, I am running a german version of Windows 10. Strange, this ist the only one dialog box within Cura with this behavior. All other dialog boxes are with german buttons.

Right now I testet it on a other machine with Windows 11 and Cura 5.1.0 fresh installed and there the buttons in the concerning dialog box are o.k. (in german).

 

I will update Cura 5.0.0 on the first machine to 5.1.0. Maybe the problem is already fixed in this version. If after the update the problem still exists then you are right and it seems I have a problem with my Windows 10. But why only this one dialog box and only Cura is affected? Never have seen this behavior with mixed languages in my other applications.

rih2555, 2022-07-25:
I updated Cura from 5.0.0 to 5.1.0 and now the buttons in the dialog box are in german. Sorry fort he circumstances. You can set the issue to solved.

closed 2022-07-26 by Vandresc