Ultimaker/Cura #12861: Consider an alternative keyboard shortcut for multiplying model

queengooborg, 2022-07-25:
### Is your feature request related to a problem?

I get very frustrated when a program overrides default shortcuts such as minimizing a window (Adobe Premiere does this too and it drives me crazy).  This is the case for the Multiply Model command, where it overrides the default command for minimizing a window on macOS.

### Describe the solution you'd like

I'd like the Multiply Model command to be assigned a different keyboard shortcut to prevent conflicts with default macOS shortcuts.  I would have suggested ⌘D originally (for "duplicate", which is what is used in Finder), however I noticed that was assigned to clearing the build plate (maybe ⇧⌘⌫ would be better for a build plate clearing shortcut).  Maybe ⇧⌘M would work here?

### Describe alternatives you've considered

N/A

### Affected users and/or printers

This will positively benefit macOS users and provide a more native-feeling experience.

Vandresc, 2022-07-25:
Hi @queengooborg , just to make sure I understand your problem, the model is selected when you use cmd + M and it still minimizes all windows? For me, it minimizes the window only when they are in focus but not when my model is selected in Cura, it correctly opens the multiply the window.

queengooborg, 2022-07-25:
It's the opposite, actually -- I WANT ⌘M to minimize the window regardless of whether a model is selected or not, as that is the OS-wide command to minimize the focused window.

Vandresc, 2022-07-25:
Aha, thank you that clear it up for me. I will bring it up for discussion with the rest of the team.

queengooborg, 2022-07-25:
To add a little bit more to this, by the way: pressing ⌘M with no model selected will do nothing, since the command is already overridden.

Vandresc, 2022-07-26:
Hi @queengooborg , we are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.
To be able to focus on the most requested and needed features from our community and from Ultimaker, we decided that we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself. 
Thank you for your feedback!

queengooborg, 2022-07-26:
I would happily open up a pull request to fix this (and attempt to fix the macOS menubar issue too), however I ran into an issue with the Conan repositories.  `cura-ce` (not the main `cura`) is requesting I authenticate, and thus I can't grab any Conan packages to set up my local environment.  Is there a specific place I should raise this issue with the Conan builds?

Vandresc, 2022-07-26:
I will ping @jellespijker, he can give you more information regarding the Conan repositories.

jellespijker, 2022-07-26:
Hi @queengooborg great initiative.
The instructions on our wiki https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source
describe how you can remove the `cura`  Artifactory repository. That one is used by Ultimaker employees and requires identification.

It could be that you'll still run in to some trouble setting things up from source. We are working on fixing this during this and the next sprint.

Let me know if you run into trouble with running from source and open up a new issue for your case or add a comment to an existing issue.

queengooborg, 2022-07-26:
Hi @jellespijker, yes I know that the `cura` repository is to be removed, which I have done.  It is NOT the `cura` repository that I am having problems with; I've already removed it.  I am having problems with the `cura-ce` repository:

```
umbase/0.1.5@ultimaker/testing: Not found in local cache, looking in remotes...
umbase/0.1.5@ultimaker/testing: Trying with 'cura-ce'...
Please log in to "cura-ce" to perform this action. Execute "conan user" command.
```

jellespijker, 2022-07-26:
Sorry I misread. We changed something on the server, which might have caused this recently. I will take a look within the hour

queengooborg, 2022-07-26:
Looks like the fix for `cura-ce` worked, it's no longer prompting me to authenticate, thanks!

closed 2023-08-09 by github-actions[bot]