Ultimaker/Cura #12862: Incorrect bottom layer pattern with spiralize outer contour (vase more)

Terones, 2022-07-25:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Anycubic Mega Zero

### Reproduction steps

Adding a model into cura, and selecting the default .2mm normal profile.
Then selecting the spiralize outer contour.
Then slice the model

### Actual results

The bottom layers are all concentric while lines was selected, even when the pattern is changed to zigzag
[image]

### Expected results

I expected the bottom layers to be lines and not concentric

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AMZ_trashcrystal.zip](https://github.com/Ultimaker/Cura/files/9181783/AMZ_trashcrystal.zip)
[image]

GregValiant, 2022-07-25:
Thanks for the report.
That is the new default and the only bottom pattern available when in vase mode is concentric.  That makes it "Expected Behavior" rather than a bug.
You aren't the first to complain about it.
I don't know where the Cura team is with this or if there are plans to address this to put it back like it was.  I'll take the "bug" label off and change this to an "Improvement" request.
Let me know if that works for you.

smartavionics, 2022-07-25:
Do you have a "regression" label, that would be appropriate.

GregValiant, 2022-07-25:
I don't see one but I'm just a greeter here.  Maybe it's in a drawer I haven't looked in yet.

Terones, 2022-07-25:
> Thanks for the report. That is the new default and the only bottom pattern available when in vase mode is concentric. That makes it "Expected Behavior" rather than a bug.

If this is indeed the expected behaviour then I expect the options int the pull-down to only show concentric.
> I don't know where the Cura team is with this or if there are plans to address this to put it back like it was. 

I like my bottom layers to be strong, having all the layers be the same pattern might compromise that, I think, though I haven't tested it thoroughly.
>I'll take the "bug" label off and change this to an "Improvement" request. Let me know if that works for you.

I'm fine with this being "improvement" as the engine has been completely changed from the previous (4.*) version.

Vandresc, 2022-07-26:
Hi all, indeed a regression. We have had a ticket on our backlog for a while now, unfortunately, the priority is not very high but still on our radar. 
Devs, see CURA-7926 for our internal reference.

dschil138, 2022-07-28:
I came here to ask this as well, I see that it's the default (only) behavior, but I do think lines should be added back as an option. The concentric pattern makes the bottom layer much less strongly attached to the vase walls. It easily detaches from the vase walls since the lines are parallel all around the edge. I just had a vase print come off the bed and immediately the base detached from ~1/3 of the perimeter walls.

GregValiant, 2022-07-28:
I agree that structurally it's a poor way to do it and for water-proofing it is also not a good idea.  Hopefully this will work it's way to the top of the stack.

mobilePilot, 2023-06-13:
I add to this request, the concentric layer doesn't work well into sticking to bed when the base layer geometry is not a perfect rectangle or circle. Please add back the possibility to do the base layer as lines or zigzag.