Ultimaker/Cura #12864: problème of slicing

lohanabr, 2022-07-25:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Wanhao D12

### Reproduction steps

i slice an piece the time it 0 and in prewiews my piece is transparent

### Actual results

the time it 0 and in prewiews my piece is transparent

### Expected results

i can print my piece 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-07-25:
Thank you for the report.
I suggest you upgrade to Cura 5.1.  There were problems with some models slicing in 5.0 and those problem models seem to slice properly using 5.1.
If you would provide a project file by loading your model, setting Cura up to slice it, and then use "File | Save Project" and zip the file and post it here someone can troubleshoot it.

lohanabr, 2022-07-26:
[rocket head v7.zip](https://github.com/Ultimaker/Cura/files/9189592/rocket.head.v7.zip)
the problem is that wanhao has not yet created profiles for cura 5.1.0

Vandresc, 2022-07-26:
It does slice for me in 5.1 but indeed 0 printing time and empty layers in preview, though I do not get the warning message in 5.1. Will be brought up for discussion with the rest of the team. Thank you!

lohanabr, 2022-07-26:
okay thanks

GregValiant, 2022-07-26:
I tried all sorts of things with that model.  NetFabb, MeshMixer, MS 3D Builder, and Formware STL repair to try to fix it. It would not slice with either IdeaMaker or PrusaSlicer although IdeaMaker did come up with this effort.
[image]

Back to Cura and turning on "Remove All Holes" I finally got something and it points to errors in the model.  None of those programs can recognize that the model is wrong and so there are no warnings or crashes, just useless slices.
This is the most useless of all.  Go through the layers from bottom to top and you will see.
Even within the gcode file layers go from 82 to 374 with nothing in between.
[image]

Ghostkeeper, 2022-07-29:
This issue is with the model, not with Cura.

Here you can see the rendering of this model in Blender:
[image]

The actual issue is that except for the bottom part (the nozzle of the rocket), this model has no width. It has an outer and an inner surface, but they almost exactly overlap. You can see that in several ways in the above screenshot:
* The edges of the triangles overlap without there being a vertex at their intersection. This is actually just two sets of triangles overlaid on each other, staggered a bit and the diagonal of each square face in the opposite direction.
* You can see some orange splotches towards the side where the two surfaces are Z fighting. This is more visible in animation than on that screenshot.

The staggered overlapping surfaces can also be seen in object mode quite clearly:
[image]

This is also what's causing the tip of the model to appear red in Cura. The "inside" surface wins out on top, turning the model inside-out. Because the normals of the inside there are facing downwards, Cura renders it as if it is overhang.

My guess is that this model is generated with CAD software that then limited the resolution for the two surfaces in different resolutions or at least staggered them, while they were originally exactly or almost exactly overlapping. Someone forgot to give the shell a width.

Removing all holes discards one of the surfaces. In places where the inner surface is actually on the inside, the outer surface is retained and this causes a disc to be sliced. In places where the inner surface is on the outside, the inner surface is retained and then proceeds to print nothing since it's a hole. Without removing all holes, in some places it will print nothing because the hole is bigger than the model, and in some places it will try printing something but that will then be too thin to print.

Since this is an issue with the model and not with Cura, we'll close this issue. It is not a bug in Cura.

closed 2022-07-29 by Ghostkeeper