Ultimaker/Cura #12868: Preheat immediately after aborting a print via usb printing causes preheat to be un-usable

RandomGgames, 2022-07-26:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Start a print via usb printing
2. Abort the print
3. While project is still aborting and ender is rehoming, mess with preheat settings (cancel and preheat)

### Actual results

Sometimes the preheat buttons never re-appear or get stuck

### Expected results

They should return to allowing preheat

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/a

GregValiant, 2022-07-27:
Thanks for the report.
Because no Ultimaker printers use USB printing it is not maintained and hasn't been for a few years.  I can understand that they don't want to spend money on something their printers don't need or use.  So it either works for you or it doesn't.
I'll leave this report open.  Maybe a member of the community will attempt to fix this.

RandomGgames, 2022-07-29:
This is a minor thing and if Ultimaker doesn't care about little things like this I wouldn't be surprised. It's a very small thing and it's not program breaking.
Maybe someone will fix it but who knows.
Thanks!

GregValiant, 2026-08-19:
There haven't been any other reports of the behavior so I'm going to close this as being stale.

closed 2026-08-19 by GregValiant