Ultimaker/Cura #12870: Erratic scroll wheel behavior

bobtato, 2022-07-27:
### Application Version

5.1.0

### Platform

MacOS 12.2.1

### Printer

Ultimaker 2+

### Reproduction steps

1. Mouse over any view with a scrollbar (e.g. the "Print settings" drop-down panel or the "About" dialog)
2. Rotate mouse scroll wheel (I am using a Logitech MX Master mouse with a physical wheel)

### Actual results

The view does not scroll, or scrolls an unusually long or short distance per wheel step.  The effect seems to vary unpredictably: for instance several turns of the mouse wheel do nothing, then the view scrolls very slowly, then a single wheel movement in the opposite direction jumps to the end of the scroll range.

### Expected results

The view scrolls by a predictable distance with each wheel event (preferably consistent with the dynamics of native UI elements).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This behavior appears whether or not a model has been loaded

Vandresc, 2022-07-28:
Devs see Cura-9382 for our internal reference.

bobtato, 2022-07-28:
If useful, I have noticed a couple of other details:

- after I scroll a view by dragging the scroll bar, any subsequent wheel event causes the view to jump to the top position

- scrolling with trackpad gestures works more or less as expected

niwla23, 2023-01-04:
Also worth noting, this issue seems to have appeared with the theme change. At least I only have it on my new machine where I have the new version of cura where everything is more spaced out for no apparent reason.
Maybe there was some change in UI Framework or something?

niwla23, 2023-01-04:
In preferences, scroll is also weird, but it does not feel like it accelerated like on mobile (as it is in print settings), but it's just extremely slow. It scrolls with half a line of text per scrollwheel click.

When going to Preferences > Settings (shouldn't it be "print settings" btw? Settings and Preferences sound kinda the same to me) it again does the accelerated scroll thing.

TLDR: scroll is broken everywhere, just in different ways

LarsStegman, 2023-01-21:
I am running into the same issue. Scrolling over the settings, it slows down the further down you go. It stops about 2/3 of the way down. If you then use the scroll bar, it resets back to the top of the list.

I am also using the Logitech MX master mouse.

aeaoeiao, 2023-01-22:
I have exactly the same issue. I'm also using MX Master 3s from Logitech.
Ironically when I use the scrolling wheel for zoom in and zoom out the buildplate I have no problems, it 'only' happens when I have to scroll between menus and preferences. It's very annoying, please fix this.

gillesguillemin, 2023-03-03:
I'd like to chime in about this issue as well. Seeing all the same behaviours described by everybody above. Also using an MX Master 3s.

gillesguillemin, 2023-03-03:
Seems related to Logitech software though as I can see the same behaviour with a (very) old Performance MX but _not_ on an equally old wired Microsoft mouse (nor with Apple Magic Mouse or Magic Trackpad, but scrolling technology is different here)

gillesguillemin, 2023-03-04:
Just realised that installing the Sidebar GUI plugin acts as a workaround, at least for the most aggravating scrolling issue, which is the sidebar scrolling (no change for the preferences but I can totally live with that.

GL17C4, 2023-08-15:
I'm having an issue with the scrolling inside Print Settings and I believe it is an issue with the mouse acceleration not being reset.
If I do multiple clicks of my scroll wheel in a row quickly, the settings page scrolls very fast.  If I then do a single click on my mouse wheel, the page scrolls at the same speed that it had from the fast scrolling.  If I do a second single click of my mouse wheel afterwards, it only scrolls a single line.
I am using different Logitech mice on two different computers with the Logitech G Hub software and both mice have the same behavior.

rob-valor, 2024-02-23:
I'm on _UltiMaker-Cura-5.6.0-linux-X64.AppImage_ and Ubuntu 23.10 and also have strange behavior of scrolling with the mouse.
It seems like Cura is _counting_ the mouse wheel _ticks_ when scrolling and scrolls the amount of _ticks_ I scrolled. So the scrolling of the pane the mouse pointer is _on_ keeps scrolling until it scrolled the amount of _ticks_ it counted when I turned the mouse wheel. But, I'm not scrolling anymore. It seems to try to mimic scrolling on a _mobile_ device (not a laptop I mean :wink: although I'm _on_ one, which makes me wonder... does that count as a mobile device for cura?)

When I scroll using the built in trackpad, scrolling is more normal. The pane scrolls while I scroll and stops scrolling when I stop. Therefor it seems to be counting the mouse wheel _ticks_. Although... when scrolling slowly with the trackpad, the pane _tries_ to scroll but is just bouncing up and down a bit. Only when I _brute scroll_, the pane start scrolling pretty fluent.

So both mouse wheel and trackpad scrolling have different issues.