Ultimaker/Cura #12871: (5.1.0) Unexpected Error on Slicing

PapaChop83, 2022-07-27:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

CR-10 V3

### Reproduction steps

Starting slice, about halfway through get this error:

"Slicing failed with an unexpected error."

[cura.log](https://github.com/Ultimaker/Cura/files/9194340/cura.log)
[Chest.zip](https://github.com/Ultimaker/Cura/files/9194356/Chest.zip)

### Actual results

I've tried other files and they slice fine, and I've run this file through Meshmixer with no issues.

### Expected results

It should slice fine.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

NA

Asterchades, 2022-07-27:
I'm unable to reproduce the issue you're experiencing, using either the profile within the 3MF (which is configured to the default Standard Quality settings) or my own profile.

With that said, the also-provided "ender3 S1 PLA p4 nozzle" profile (which wasn't selected) does produce the issue. The root cause seems to be the "Connect Infill Lines" setting - disabling that allows it to slice properly. I know this setting was an issue during the beta  (#11141) but was under the impression it was resolved before release, so either this is an edge-case model or there's been a regression.

_For ease of diagnostics, in future it would be beneficial to ensure your project is saved while in the state being reported. As saved this file had no issues._

Vandresc, 2022-07-28:
Hi @PapaChop83 , exactly as @Asterchades explained, we cannot reproduce your issue with the default Standard quality. Could you pls upload a project file with the state that the slicing error occurs? Thank you!

Sineos, 2022-07-29:
The mesh has massive errors: holes, manifold triangles, boundary edges. Maybe your mesh-fixing settings are not aggressive enough. 

Try:
[beskar_chest_whole.stl (repaired).zip](https://github.com/Ultimaker/Cura/files/9216545/beskar_chest_whole.stl.repaired.zip)

closed 2022-08-11 by github-actions[bot]