Ultimaker/Cura #12872: Slicing failed with an unexpected error

ultrasuperman1001, 2022-07-27:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

I imported the attached project file and tried to slice with my usual settings. When supports are selected the progress bar doesn't advance at all, when supports are enabled, it will progress about 10% then the error "slicing failed with an unknown error".

The slicing works just fine in Cura 4.13.
[durango error.zip](https://github.com/Ultimaker/Cura/files/9194391/durango.error.zip)
1

### Actual results

an error message will pop up

### Expected results

the program should slice

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

the slicing works fine in 4.13.1

Asterchades, 2022-07-27:
Immediately upon loading this model it's clear it needs serious repairs. A warning is presented stating that the model is not manifold and you're immediately presented with a preview covered in the magenta/cyan polka dot pattern indicating all the non-manifold areas. Further to this if you enable the X-ray view you can see large amounts of the model where odd numbers of walls are present, marked in red.

Despite that, it can still be sliced. It cannot be sliced _well_ but it can be done.

The issue seems to stem from the "Keep Disconnected Faces" mesh fix. Disabling that lets it slice, and enabling it even from the default Standard Quality profile will stop it. However it is not that setting in isolation as using my own profile I'm able to enable it and slice without issue (beyond the obvious issues in the resulting slice due to the model geometry). Further, adjusting the scale while using the Ender profile will also allow that setting to be used.

This may or may not be related -at least in part - to #12771 where specific geometric circumstances could result in a crash. However the state of the model will make diagnostics extremely difficult. I'm unable to confirm whether a repaired version of the model still suffers the issue as the various repair tools I have tried have all produced results visually different from the input (such as completely sealing the cabin).

MariMakes, 2022-08-25:
Hey @ultrasuperman1001,

@Asterchades is right. This model is not made for 3D printing. 
Your object is not watertight and consists out of too many sub meshes. 

If this is your model, please consider simplifying the amount of submeshes, make sure it's watertight and export is as an .STL or .3mf.  OBJ files are more commonly used in games and are not made for 3D printing. The OBJ file is just the container for the data, not a guarantee that something is printable. Keep in mind that OBJ files are also used as an interchange format for 3D graphics applications, a use case that has nothing to do with 3D printing. 3D models suitable for 3D printing need to have a minimum wall thickness and a “watertight” surface geometry to be 3D printable. 

If you are in doubt, if this was a good model you would see no discolorations in your preview. 
[image]
[image]

I'll be closing this issue since it's not something that we can solve with Cura.

closed 2022-08-25 by MariMakes