Ultimaker/Cura #12874: Incorrect Keepout Using Build Plate Adhesion w/ Multiple Extruders

muhcabbages, 2022-07-27:
### Application Version

5.1.0

### Platform

windows 10

### Printer

custom FFF printer

### Reproduction steps

Using a modified Creality Ender 3 w/ a second hotend mounted left and forward of the original printhead. Extruder 1 can reach the entire print bed (235mm square) and extruder 2 can reach from origin to 220,235.

1. under Settings > Printer > Manage Printers, hit Add New
2. select non-networked, Custom FFF printer
3. set build area to 235 x 235 x 210 w/ heated bed
4. set Number of Extruders to 2
5. under Extruder 2 tab, set offset to -35.5 x -5
6. begin configuring a print with this printer
7. set Build Plate Adhesion to Brim (note that Extruder 1 is selected by default)

### Actual results

The keepout box prevents models from being placed within ~40mm of the left edge of the build plate, or ~10mm of the front edge. The box is being placed as though Extruder 2 were homed to 0,0 with Extruder 1 having a positive offset relative to Extruder 2. This is contrary to configuration.

[image]

### Expected results

Based on the configuration features available and the settings as entered, the keepout box should be keeping models away from the right edge of the build plate rather than the left edge.

For the keepout box to give CORRECT results for my printer, it's necessary to be able to configure a maximum X travel greater than the X size of the print volume.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There _**NEEDS**_ to be a big checkbox to ignore the keepout box. The idea that y'all are trying to make it "aware" of multiple extruders when it's not possible in the software to fully specify the parameters of a multi-extruder printer really irks me.

GregValiant, 2022-07-28:
Thanks for the report.
Cura has printer definitions for over 300 printers.  You have a 2nd party printer customized with 3rd party parts that were installed by a 4th party.  Once you start customizing something you are going to run into compatibility issues.  Everything affects everything.
I don't work for UM and I don't know for sure, but I have doubts that the Ultimaker 3D Printer Manufacturing Company will spend their money to develop special software specifically for a 1-off custom Creality printer.
Have you made out a bug report for the Creality slicer?  It's based on Cura and put out by the people that built your printer.  (They also built my printer which is an Ender 3 Pro.)

I believe you will be looking at workarounds and here are some things you can try.
Increase the size of the build plate to 270 x 240.  You won't be able to use the extra 35mm on the left as it will be off your "real" build plate.  Cura should stick the disallowed area there because extruder 1 can't get there.
You might have to adjust your home offset using M206.
You could adjust the size and location of the Creality_ender3.3mf base mesh so that it is properly located within the Cura workspace and properly shows the true relationship to the right extruder#1.
That could get you something like this:
[image]
Both models are set to E2.  The scored line on the left represents the edge of the real bed and is the farthest a model can go since E1 can't travel into that area.  The gray area on the right is the disallowed area since E2 can't get into that area.  That leaves just about 200mm between the "scored" line on the build plate and the disallowed area on the right.  That is the range of the left extruder.
This is with both models by E1.  The range of E1 is the full 235mm of the build plate.
[image]

This next image is with each model set to print with a different extruder.  You can see that everything to the left of the scored line is gray and disallowed but in reality that area is off the build plate.
[image]

I really don't see a problem accommodating your printer with a little knowledge on how things work in 3D printing and fudging some numbers in Cura.  I can't check because I don't have a printer like yours.
I'll leave this as a bug report and someone from the Cura team will take a look.  Maybe the program isn't interpreting the negative offset values correctly, or maybe Cura just always assumes that neither extruder can move outside the bed.

muhcabbages, 2022-07-28:
Thinking of 3D printers as being "from a manufacturer" fails to appreciate how completely modular and generic these devices are. If anyone is producing a slicer software based on open-source tech and they are working exclusively from the perspective of supporting unmodified commercial machines, frankly they are failing the spirit of the community. I've never gotten the vibe that Ultimaker is that brand.

No I have not entered a bug report with Creality's slicer because I do not use it and frankly was not aware that it exists. My printer is not remotely in-spec for a Creality printer. It has a BTT Octopus motherboard populated with 2209s, dual Z axis, sensorless X homing, etc. Even when it was in-spec for an Ender 3 Pro, I haven't run it based on somebody else's presets since my first handful of prints. The presets available in these slicers are a starting point AT BEST - and all of this is totally beside the point of what a slicing software suite IS and what it's supposed to support.

For the record, this was a bug report and not a support request. My mind has not been blown by the revelation that I can cheat the keepout square by lying about the bed size. I can also do my large prints with Bed Adhesion set to None as the bug seems to exist specifically in that feature.

Ghostkeeper, 2022-08-01:
Cura is also very modular and extensible. It can accommodate this set-up, but it requires some finagling because it's quite rare.

Cura can make one of two assumptions, which can be switched between depending on the `nozzle_offsetting_for_disallowed_areas` machine setting:

* By default, Cura will assume that the reach of the gantry is the limiting factor for the build volume. With an offset of 35mm between the nozzles, that means that the left nozzle can reach 35mm further left than the right nozzle, and the right nozzle can reach 35mm further right than the left nozzle. This is what you're seeing here. It means that there will be a disallowed area on the left when using only the right nozzle, a disallowed area right when using only the left nozzle, or both when using both nozzles.
* If `nozzle_offsetting_for_disallowed_areas` is disabled, Cura assumes that the build plate is the limiting factor. Both nozzles can only print on the build plate, and so there is no disallowed area on the sides.

So to answer this request:

> There NEEDS to be a big checkbox to ignore the keepout box.

There is an option for it, but it's rare enough that we didn't make a GUI checkbox for it. To use it, adjust the printer definition to set `nozzle_offsetting_for_disallowed_areas` to `false`.

Your desired behaviour is actually that the reachable volume is limited by the gantry on the right-hand side, but limited by the build plate on the left-hand side. There is no automatic option to get that behaviour in Cura, but it can be spoofed. To get the precise behaviour, you need to adjust the printer definition like GregValiant suggested:

* Increase the size of the build plate to include 35mm on the left, to 270mm.
* Add M206 to adjust the zero coordinate in the start g-code.
* Add an extra disallowed area as polygon to the definition file in the `machine_disallowed_areas` setting.
* Make sure that `nozzle_offsetting_for_disallowed_areas` is enabled (as is default) to make sure that you still get the proper disallowed area on the right.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/10354, which was really the same problem. See that issue for more detailed analysis.

closed 2022-08-01 by Ghostkeeper

muhcabbages, 2022-08-10:
If it was simply possible to configure the safe limits of movement for each axis it would become very easy to support this behavior completely and correctly for all Custom FFF rigs through the GUI.

I do not understand your justification for marking these issues as Closed. The existence of a workaround does not close a bug ticket. The way the keepout rectangle interprets the hotend position configuration _is _incorrect__ and the _least_ hacky remedy somehow is to deactivate the feature entirely through a configuration option documented only by forum posts and bug tickets. Y'all even quietly altered the location of the printer definition files in Cura 5.

Here's another reason why I feel this option should be available in the GUI (and why I am choosing to disable the rectangle for my own convenience) - it simply doesn't understand that Bed Adhesion features are only generated around parts of the model that touch the build plate 
[image]

If there's no intent to fix of any of this, could we at LEAST get a link to some proper support documentation in the Manage Printers menu?