Ultimaker/Cura #12876: Slicing Failed

Andrethetech, 2022-07-27:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Custom CoreXY

### Reproduction steps

Try slicing acceleration tower from parts for calibration

### Actual results

Slicing failed

### Expected results

Slicing working , this worked in the previous version.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura 5.1 beta acceleration tower.zip](https://github.com/Ultimaker/Cura/files/9197334/Cura.5.1.beta.acceleration.tower.zip)
[cura.zip](https://github.com/Ultimaker/Cura/files/9197335/cura.zip)

Vandresc, 2022-07-29:
Thank you @Andrethetech, we now have a ticket on our backlog. Devs see CURA-9380 for our internal reference.

rburema, 2022-08-19:
Tested the initial .zip the OP posted on an early alpha build of 5.2 -- slicing succeeds now. Closing this. If this (eventually, when it's out) still occurs in 5.2 beta or later, please open a new ticket (or ask us to reopen this one).

closed 2022-08-19 by rburema