Ultimaker/Cura #12877: ( 5.1.0 ) Unable to open File ( STL )

TwomD, 2022-07-27:
### Application Version

5.1.0

### Platform

win 10

### Printer

ender 3 pro

### Reproduction steps

since updating to cura 5.1.0 i cant load any stl files i generate myself.
it always shows the error message: "Unable to open File, The File could be corrupt, inaccessible or it did not contain any models"
cura 5.0.0 ( and any before that ) open them just fine ( so does windows builder and any other program i tried ) without any problems.

the stl was created using "Megacad Profi Plus 2017". I  included a simple stl in a zip file so you can check yourself.

### Actual results

cant open stl files

### Expected results

opening stl files ;)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Mike Kamerahalterung.zip](https://github.com/Ultimaker/Cura/files/9199291/Mike.Kamerahalterung.zip)
[image]

GregValiant, 2022-07-28:
Thanks for the report.
I agree that something is going on with that model.  I was able to open it in 4.12.1 but not with 5.1.  It doesn't show any errors in MS 3D Builder.
I was able to export the STL from 4.12.1 and  then open that new STL in 5.1.
Somebody from the Cura Team will take a look.

Vandresc, 2022-07-29:
Hi @TwomD, the issue lies somewhere with numpy-stl library, in Cura 5.0 due to a build system issue this Python package was not packaged in the application, and it was still using a fallback, that is the reason that it was still working in the build before 5.1. We did fix the packaging issue and numpy-stl is used correctly again but it fails to load certain models, like your case.
As there are some workarounds, see @GregValiant comment, we agreed to defer this issue for now.
Thank you for bringing this to our attention!