Ultimaker/Cura #12883: STL loading issue

Cramster, 2022-07-27:
### Application Version

5.1

### Platform

macOS 11.6.7

### Printer

Reality Ender 3

### Reproduction steps

Load the attached STL and then attempt to slice it, the program consistently fails. This is the only STL I've ever seen fail, out of many.

### Actual results

Cura shows a slicing error with an OK button

### Expected results

I would expect the STL to slice, prepping it for printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It won't let me attach the STL but here it is on this thingiverse page, it is the only STL available to download for this file: https://www.thingiverse.com/thing:4575001/files

Cramster, 2022-07-27:
[image]

GregValiant, 2022-07-28:
Thanks for the report.
When you loaded the STL did you get the "Not Watertight and may not print correctly" warning?  There are errors in the model and so it won't slice as it is.  You can use MS 3D Builder or a free on-line service to repair the model.
Here is a report on the errors in the model.
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 1 Non-manifold edges (?)
--> 1 Inverted faces (?)
--> 22 Degenerate faces (?)**
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 124012ms ------
-> Vertex count changed from 23117 to 56831 **(+33714)**
-> Triangle count changed from 51782 to 129550 **(+77768)**
This is a model issue so I will close the report.  You can still respond here if you think that is an error.
Here it is sliced after being repaired.
[image]

closed 2022-07-28 by GregValiant

Cramster, 2022-07-28:
Thank you for this info, I will attempt to repair the stl file. I did not receive a "Not Watertight and may not print correctly" warning.

GregValiant, 2022-07-28:
Go to the Cura MarketPlace and load the "Mesh Tools" plugin.  You will have to quit Cura and restart it.
It has some handy tools for dealing with models.  I don't think it would have repaired this one, but it works on a lot of them.

Cramster, 2022-07-28:
Thanks for the follow-up, I'm going to install that now. Also I got the model working by following the steps in this brief Mesh Labs tutorial: https://www.youtube.com/watch?v=IreaoCfQ8mE
I previously tried this free web tool (formware.co/OnlineStlRepair), which found the issues you did (non-manifold edges, etc.) but the 'fixed' stl it produced did not appear to work still.

GregValiant, 2022-07-29:
It's odd how some models just have a lot of errors.  I guess it depends on the software that created them.  Some of the character figurines are really bad.  I don't know if they were scanned or drawn from cartoons but they seem to have the most errors.  There was a bug report here and the model had over 400,000 errors.  I think there were more things wrong than right.
I design most of my own stuff but when I do grab an STL from someplace like Thingiverse I always check it for errors first thing.  It can save me a lot of head scratching later.

Cramster, 2022-07-29:
For sure, I'm still fairly to printing but one of my goals is to work more with software like Mesh Labs to try to create/edit my own stls. ^^