Ultimaker/Cura #12884: Storage Device File Sorting by Material or Similar

Black-Villain, 2022-07-27:
### Is your feature request related to a problem?

I get kinda frustrated when sorting through files on my printer or remembering what exact print my file is for. For example, I use one SD card for my two identical printers, have for a couple of years, so I have dozens if not a 1-200 print files on this SD card. When I want to print an old file on that card, I kinda have to guess or try to remember what material I set that file up for.

### Describe the solution you'd like

I'd like a feature to either allow Cura to make folders on your storage device/SD card, or add a suffix to the filename, to denote what material profile is selected in Cura for that print job. For example, if I have a file named "Container" in Cura, I'd like for Cura to be able to automatically rename that file to "Container_ABS" or "Container_PETG" _or_ allow Cura to make folders on the device to sort print jobs/gcodes automatically like:

Root Storage
├── PETG
│   ├── File1.gcode
│   ├── File2.gcode
│   └── File3.gcode
└── ABS
    ├── File1.gcode
    └── File2.gcode
OR
└── PETG
    ├── Generic
    │   ├── File1.gcode
    │   └── File2.gcode
    ├── 3D-Fuel
    │   └── File1.gcode
    └── eSun
        └── File1.gcode

### Describe alternatives you've considered

- I've renamed the files manually after Cura saves them locally or to the storage device (annoying to do every time and time consuming if I'm setting up multiple print jobs/parts/assemblies)
- I've made folders on the storage device like above, but again I have to manually select the file save location every time, which gets annoying when I'm setting up multiple jobs or if I just forget

### Affected users and/or printers

I think this could benefit a lot of users, especially those dealing with creating a large number or print files, or users that naturally have a large number of files accumulate over time like myself.

Vandresc, 2022-08-02:
Hi @Black-Villain, we are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.
To be able to focus on the most requested and needed features from our community and from Ultimaker, we believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

jellespijker, 2022-08-02:
The Ultimaker printers won't benefit from such a changes. But this would be a great plugin imo.

Maybe someone in the community will pick this up.

closed 2023-08-17 by github-actions[bot]

fieldOfView, 2023-08-17:
For the benefit of those finding this issue via a search engine: You may be interested in the Printjob Naming plugin available from the Marketplace. It lets you set a custom name pattern for the gcode file itself AND for the path (subfolders).