Ultimaker/Cura #12885: [5.0] Draft Shield w/Dual Extruders, Order of Operations Incorrect

rachael7, 2022-07-27:
**Application Version**
Cura 5.0

**Platform**
Windows 10

**Display Driver**
NVIDIA Quadro RTX 4000, Driver version 31.0.15.1625

**Steps to Reproduce**
1. Slice any dual extruder print, such as the bi-color cube from the Parts for Calibration plug-in, including a draft shield.
2. Use the playback on the preview screen to see the order of operations.

**Actual Results**

With a dual extruder machine, Cura prints two layers with each extruder, to save on extruder switching. When a new extruder is selected, it needs to be primed, but Cura has the new extruder first finish its portion of the layer currently printing, before printing the draft shield. Therefore, the extruder starts printing before being primed. The order of operations, for any arbitrary sequence of layers after the first layer, currently looks like this:

Layer 3 (arbitrarily chosen): Extruder 1, still selected and primed from layer 2, prints draft shield layer 3, then prints its portion of part layer 3. The extruders then switch and extruder 2 prints its portion of part layer 3 (with no prior priming).
Layer 4: Extruder 2, still active, prints draft shield layer 4, then its portion of part layer 4. Extruder 1 then prints its portion of part layer 4 (with no prior priming).
Layer 5: Extruder 1, still active, prints draft shield layer 5, then its portion of part layer 5. Extruder 2 then prints its portion of part layer 5 (with no prime).
Layer 6: Extruder 2, still active, prints draft shield layer 6, then its portion of part layer 6. Extruder 1 then prints its portion of part layer 6 (with no prime).

**Expected results**

According to the settings guide, the draft shield can be used to prime the extruder in a multi-extruder print. But in order to serve that function, the draft shield would have to be the first thing printed after an extruder change, rather than the first thing printed on a new layer, as it is presently. I submitted a bug report for the settings guide plug-in to alert them to this error. They responded that Cura is supposed to print the draft shield first after an extruder change, to serve as an extruder primer, and its failure to do so is a bug. They continued to say that the settings guide does not document bugs, but rather only the expected behavior, so they would not update the settings guide and instead a bug report should be submitted for CuraEngine (see [Settings Guide Issue 92](https://github.com/Ghostkeeper/SettingsGuide/issues/92)). This is that report. Here is an alternate order of operations which would serve the purpose of using the draft shield to prime the extruders, without violating the prohibition against moving down to print earlier layers (which would risk collision):

Layer 1: Extruder 1 prints its portion of bed adhesion (skirt, brim, raft, prime blob), which serves to prime the extruder, then prints its portion of part layer 1. Extruder 2 prints its portion of bed adhesion, then draft shield layer 1, either or both of which prime extruder 2, then extruder 2 prints its portion of part layer 1.
Layer 2: Extruder 2, still active and primed, prints its portion of part layer 2. Extruders swap and Extruder 1 prints draft shield layer 2, priming the extruder in the process, then prints its portion of part layer 2.
Layer 3: Extruder 1, still active and primed, prints its portion of part layer 3. Extruder 2 then prints draft shield layer 3 (priming the extruder), then prints its portion of part layer 3.
Layer 4: Extruder 2, still active and primed, prints its portion of part layer 4. Extruder 1 prints draft shield layer 4, then its portion of part layer 4.

With this order of operations, each extruder always gets an opportunity to prime by first printing the draft shield and the cycle keeps all printing to only one layer at a time, avoiding any possibility of collision with the previously printed part.

**Additional Information**

This same cycle change would also work with the ooze shield as well, allowing it to serve the purpose of priming the extruders, in addition to its other functions. As it stands now, to get proper extruder priming, one must use a prime tower in addition to the ooze/draft shield, wasting time and material. This simple change in the order of operations would greatly increase the utility of both the ooze shield and the draft shield for all dual extruder users, including Ultimaker S3 and S5 users. The draft and ooze shields are not commonly used, but this change should be simple enough to be justifiable for those who need to use those functions and the draft/ooze shields might see much more use if they could also replace the prime tower (which is commonly used).

An alternate solution would be to have the ooze/draft shield comprised of one loop from each extruder, the way the prime tower is done. That would allow each extruder to prime on the shield, it would strengthen the ooze/draft shield, and it would improve adhesion in the ooze/draft shield, since each material would print on top of the same material in the shield, rather than alternating every other layer as it does now. This solution would use more material and take more time, but that might be offset in some cases by the sturdier shield and improved adhesion within the shield (and associated reliability). An ideal solution might incorporate the first change I suggested above, with a checkbox option for a two-wall shield using both extruders.

jellespijker, 2022-07-27:
@Ghostkeeper you mention that it is on the backlog can you add the relevant Jira ticket number?

Erin-Stephan, 2023-05-11:
I am still having this issue on Cura 5.3.0. Is someone trying to resolve this? I essentially have to use the prime tower always because the ooze and draft shields are not in the right order to use as dual extruder priming.
tagging @jellespijker and @Ghostkeeper since you both are the last to interact with this ticket. Thank you