Ultimaker/Cura #12888: Plugin Side Bar

Jaroslav99, 2022-07-28:
### Application Version

5.1.0

### Platform

windows10

### Printer

CR10S-PRO V2

### Reproduction steps

Plugin Sidebar GUI 4.3.0 not work

### Actual results

[image]

### Expected results

Fix plugin

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

not

fieldOfView, 2022-07-28:
Normally, issues for plugins should go to the issue queues of the respective plugin, not to the issue queue of Cura. The creators of Cura don't maintain the plugins,

In this case, that issue queue is here: 
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues
More specifically, you will find this pinned issue on the subject which also contains a note about Cura 5.1:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/31

Having said all that, I have pushed a new version of the plugin to the Marketplace early last week, but it is awaiting approval. This new version adds support for Cura 5.1. I'll leave this issue open for Ultimaker devs to find, as a subtle reminder that my plugin needs a review.

jellespijker, 2022-07-29:
Hahaha 🤣 I love subtle reminders, it was on my todo list for yesterday, but I got distracted (oeh a 🐿️). I will pick this up first thing today 😉

jellespijker, 2022-07-29:
Plugin is reviewed and Approved. I will close this issue as a subtle reminder for @fieldOfView to now publish the plugin on our marketkplace

closed 2022-07-29 by jellespijker

fieldOfView, 2022-07-29:
The new version is available in the Marketplace now. Get it while it's hot.
Thanks for the review @jellespijker !

Jaroslav99, 2022-07-29:
Hello Aldo,

Thanks for help. Have a nice day.

-- 
Best regards,
 Modelkit                           ***@***.***