Ultimaker/Cura #12889: Unable to Slice/Temp. not reading correctly

FacelessPotatoPie, 2022-07-28:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Put my settings into Cura
Hit Slice
Error: Unable to slice.

Second issue:
Settings are correct in Cura but when a print starts, the hotend temp drops to 175.  This started happening when I updated to 5.1.0

### Actual results

First Issue: Receive error unable to slice

Second issue: Hotend temp drops to 175 and have to correct it in printer settings.

### Expected results

Issue 1: Should be able to slice object

Issue 2: Temp should not be dropping

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional information is present.

GregValiant, 2022-07-28:
Thanks for the report.
Load a model you are having trouble with and set it up to slice.  Use the "File | Save Project" command to create a 3mf project file.  You will have to Zip the file to post it here.  It's needed to troubleshoot your issue.

Asterchades, 2022-07-29:
Regarding issue 2: How long are you waiting after seeing the temperature being set to 175°c before manually adjusting?

The Ender-3 Pro start GCode heats the nozzle to the material _standby_ temperature first - not the initial layer temperature. It will then go on to heat the bed (waiting for temperature), followed by heating the nozzle the initial print temperature (also waiting). In other words it _should_ be set to 175°c for many materials and will make a correction later without your intervention.

This was a deliberate change to the start code that was merged on December 24 last year. If you feel it should be different, I would encourage you to make a pull request with your proposed changes.

FacelessPotatoPie, 2022-07-29:
Hello, Thank you for getting back to me.  This is the file I'm having issues with in regards to not being able to slice.  Thank you for explaining the temp issue to me.  I'll keep an eye on it and see if it continues, and if so, if it changes at all.  
[CE3PRO_3dBaby_onepiece.zip](https://github.com/Ultimaker/Cura/files/9223909/CE3PRO_3dBaby_onepiece.zip)

GregValiant, 2022-07-30:
Something must have sliced because that is a gcode file and not a project file.  Gcode files can tell us where you went but not how you got there.
That file will fail because there is no support inside the head and the right eye starts printing over air.