Ultimaker/Cura #12890: Printing Stops Mid-Print

whitecheddar, 2022-07-28:
### Application Version

5.1.0

### Platform

Windows 11 & Nvidia GeForce GTX 1060

### Printer

Longer LK4 Pro

### Reproduction steps

Settings have remained constant. 

### Actual results

Some prints fail several hours in, some fail with only 5 minutes left. There seems to be no pattern. Or at least no pattern that I can see. Cura is on latest version (to my knowledge) and printer is on latest version. However this problem occurred before updating both to current versions as well.

### Expected results

A successful print. I've looked online and it seems this has been a cura problem since at least 2018. Please get this fixed. It's shocking that it's lasted this long.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[output_log.txt](https://github.com/Ultimaker/Cura/files/9212695/output_log.txt)

GregValiant, 2022-07-28:
Thanks for the report.
Since no Ultimaker printers use USB printing it is unmaintained by the UM Cura Team so unless a member of the "community" wants to take on the project it will likely continue to have issues. 
There are a lot of different USB configurations on printers from multiple manufacturers and a lot of different port drivers on computers from a lot of different manufacturers.
In a previous life I did a bit of work with software for communicating with serial devices (data collection mostly).  After looking at 3D printing over the USB ... even if I was capable I wouldn't want any part of it.

Cura (and Pronterface) had an issue with print stoppages (when USB printing) but there was always "click to resume..." on the LCD and a button click would get it going again.  The new thing that seems to be happening is that the computer loses the printer port for some reason.  I have not seen a good answer as to why it happens.

I'll leave this as a bug report.  Your best options are to use the SD card (simple and works well) or move to OctoPrint/RaspPII (more complicated but seems to work well).

dertobias, 2022-11-17:
Hi! have the same problem since 5.0.0. also with 5.2.1
USB connection suddenly disappears on printer, randomly. never had an issue before.

GregValiant, 2022-11-17:
My answer above still applies.
Until someone figures out what is going on it probably can't be fixed.  Even if that happens, the question of who is going to fix it will remain.

dertobias, 2022-11-22:
switched to octoprint on a rasperry pi  and the octoprint plugin in cura
now i'm able to print [longer] again (testet a 4h print today).