Ultimaker/Cura #12891: Travel Move Causes Defect On Wall Of Project

MiSTerConsoles, 2022-07-28:
### Application Version

5.1.0

### Platform

Windows 10 Pro, NVIDIA GeForce GT710

### Printer

Creality CR-10S Pro V2

### Reproduction steps

1. Load STL files into Cura for top and bottom of an enclosure/box
2. Slice each part separately with Cura
3. Save output files for each component
4. Print output files for each component

### Actual results

When printing both halves of the enclosure, a travel move outside the box on layer 104 (same for each part) causes a string and blob to appear on the left and right walls. 
[image]
[image]
[image]

### Expected results

With Combing set to "All", I expected the nozzle to not travel across walls which would prevent this defect on the outer skin. It does not occur on any other layers for the project.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Project File.zip](https://github.com/Ultimaker/Cura/files/9213586/Project.File.zip)

MariMakes, 2022-08-25:
Hey @MiSTerConsoles,

Thanks for your wonderfully clear report 🎉 
You are totally right, this looks like a bug.
The behavior improves if I change the combing [to Within Infill](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_combing.md) but this still looks like unintended behavior. 

We already had a internal report to something similar, so I'll add it as a usecase.
For internal reference CURA-9086  
🤞 Fingers Crossed

Piezoid, 2022-08-25:
Possibly related: #13061

MiSTerConsoles, 2022-08-25:
Thanks @MariMakes and @Piezoid for looking into this issue. I will try the [to Within Infill](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_combing.md) setting or maybe even go back to the latest version 4 release when I print another one.