Ultimaker/Cura #12892: Cura 5.x keeps forgetting save and load location

robertkardos, 2022-07-29:
### Application Version

5.x

### Platform

Linux, Ubuntu 22.04

### Printer

Ender 3V2

### Reproduction steps

- Open Cura 5.x appimage
- Open a project file somewhere not in the folder that just opened by default
- Try to either save project or save .gcode file

### Actual results

Save dialog does not open at the location of the project file but probably at the user home folder.

Also, save dialog opens up behind Cura window, throwing a notification titled "Portal", with the text "Save to Disk is ready".
Might be related, might not be, I thought I'll mention it just to be sure. This does not happen in Cura4.x versions.

### Expected results

In Cura 4.x when I try to save, it automatically opens up the last folder I was working within.
Probably this should also happen in Cura 5.x too.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Log after startup [cura.log](https://github.com/Ultimaker/Cura/files/9215229/cura.log)
Log after saving the testfile to the default folder which is my home folder [cura_after-saving-testfile-to-default-folder.log](https://github.com/Ultimaker/Cura/files/9215264/cura_after-saving-testfile-to-default-folder.log)
Log after saving the testfile into a specific folder [cura_after-saving-testfile-to-testfolder.log](https://github.com/Ultimaker/Cura/files/9215263/cura_after-saving-testfile-to-testfolder.log)

After this my cura.cfg contains stuff like this:

```
[local_file]
dialog_load_path = 
last_used_type = application/vnd.ms-package.3dmanufacturing-3dmodel+xml
dialog_save_path = /home/krobz
```
So somehow the dialog_save_path variable did not get modified even though in my "recent_files" variable does contain the info:
```
[cura]
active_setting_visibility_preset = custom
active_mode = 1
expanded_brands = ;Generic
recent_files = /home/krobz/TESTFOLDER/TESTFILEINTESTFOLDER;/home/krobz/TESTFILE;/home/...
custom_visible_settings = machine_settings;speed_travel;acceleration_enabled;supp...
active_machine = E3v2_pimped
categories_expanded = ;acceleration_enabled;acceleration_infill;acceleration_ironing;accele...
jobname_prefix = False
dialog_on_project_save = False
```

ramonskie, 2022-07-31:
im runnning cura 5.1-beta and have the same problem.
even when i set `dialog_load_path =`
it just reset `dialog_load_path` top empty on restart

Vandresc, 2022-08-01:
Hi, we have a ticket on our backlog related to this issue. I will close this one as a duplicate of https://github.com/Ultimaker/Cura/issues/12295, pls follow the progress in there. Thank you for your feedback!

closed 2022-08-01 by Vandresc