Ultimaker/Cura #12894: Flexi_Fokobot slicing failing @ .16 layer height

phoenixfire425, 2022-07-29:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 5 Plus

### Reproduction steps

Used this STL

https://www.thingiverse.com/thing:5152388

Set the layer height to .16 and sliced.

### Actual results

Slicing error and cannot export the gcode

### Expected results

ability to export and print the gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9220505/cura.log)
[image]
[E5P_Flexi_Fokobot.zip](https://github.com/Ultimaker/Cura/files/9220519/E5P_Flexi_Fokobot.zip)

phoenixfire425, 2022-07-29:
Slicing in 5.0.0 works fine.  this only occurs in 5.1.0

GregValiant, 2022-07-30:
Thanks for the report.
I'm running Win10 and it sliced for me in Cura 5.1 so I am unable to duplicate the problem.
To troubleshoot this you will need to post a project file and the Cura log file.  The log file can be found at "Help | Show Configuration Folder" and is named "Cura.log"
For a project file - load the model and use the "File | Save Project" command to create a 3mf file.
Put both files into a ZIP Folder and post the ZIP folder here.