Ultimaker/Cura #12896: not loading models in from a saved file when opening

GaryLesterLMT, 2022-07-29:
### Application Version

5.0.0

### Platform

Windows 11 pro

### Printer

ultimaker S5

### Reproduction steps

1. inserted .stl into build area
2. set parameters on right
3. done with file and now saving
4. saved project to disk
5. come back later on to open file
6. file start loading in and shows status bar down at bottom that its loading
7. then status bar goes away
8. then screen is blank like as if you start started the software

### Actual results

screen is black and file/program is not loaded

### Expected results

file/program should have loaded with the .stl and the parameters

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9220997/cura.log)

Asterchades, 2022-07-30:
Diagnostics cannot be done without you also providing an affected project file. Other projects are known to load both the models and profiles without issue as they're used for diagnostic purposes.

fieldOfView, 2022-07-30:
In addition to what @Asterchades said, please include the full Cura.log, instead of an excerpt. For example in this case it would be interesting to see what happened when _saving_ the project, not just when loading the project,

GaryLesterLMT, 2022-08-11:
I have attached the full log file and the final project
[cura.log](https://github.com/Ultimaker/Cura/files/9310019/cura.log)
.
Let me know if this helps anymore. github says my file size is too big still even when zipped.

closed 2022-08-11 by GaryLesterLMT