Ultimaker/Cura #12897: Ultimaker slicing

BaronVonJ, 2022-07-29:
### Application Version

5.1.0

### Platform

Mac

### Printer

Creality Ender3

### Reproduction steps

Loaded a file. tried to slice

### Actual results

error: slicing failed with unexpected error. please report. Doesn't matter what model I load.

### Expected results

sliced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-07-29:
Thanks for the report.
The Cura team will need a couple of things to debug this.
Under the "Help" menu is "Show Configuration Folder".  Click on that and you will see the "Cura.Log" file.  Add the Cura.log file to a new zip folder (right click on it and choose "Send to" and "Compressed (zip) folder")>

With the model loaded and Cura ready to slice select the "File | Save Project" command.  Add that new 3mf file to the zip folder and post the Zip folder here.  Without those files it can't really be investigated.

fieldOfView, 2022-07-30:
> ... and post the Zip folder here. Without those files it can't really be investigated.

NB: sending in the requested files as an email attachment will NOT work. Please use the github web interface instead:
[github.com/Ultimaker/Cura/issues/12897](https://github.com/Ultimaker/Cura/issues/12897)

BaronVonJ, 2022-08-01:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/9234037/cura.log.zip)

MariMakes, 2022-08-25:
Hey @BaronVonJ,

Welcome to Github 🎉 
Thank you for opening your first issue with us 😄 

I'm looking at your logs and I see 
_Reversing even wall line causes it to be printed CCW instead of CW._  🤔 

It would really help if you could share your project file as @GregValiant suggested: 
**With the model loaded and Cura ready to slice select the "File | Save Project" command. Add that new 3mf file to the zip folder and post the Zip folder here. since your current zip only contains your cura log file** 

Without those files it can't really be investigated. 😢