Ultimaker/Cura #12899: Exact Z Seam Positions

tstone52, 2022-07-29:
### Is your feature request related to a problem?

I cannot put the z seam in an exact custom position. I can adjust the x or y position of the z seam, but if it is a curved model, the actual z seam is not straight up in down. It zig zags back and forth along the contours of the curved surface.

Many models I print for commercial purposes require a straight up and down z seam in a precise location for aesthetic purposes. I cannot achieve this on Cura.

### Describe the solution you'd like

A community member has released a modified version of Cura that has this feature. I love it. It works great. Unfortunately it is on Cura 4.10 which is now outdated. I would like to be able to use this feature with the newer Cura because I want to also use Arachne.

The custom version of Cura with this feature is detailed in this thread. There is a link to it in the thread as well.
https://community.ultimaker.com/topic/29901-testers-required-for-exact-z-seam-positioning-in-cura-development-builds/?_fromLogin=1#replyForm

### Describe alternatives you've considered

I considered using Prusaslicer because it also has Arachne, but the seam positioning is not so precise, even with their "paint-on seam" feature. I also really like tree supports in Cura and Prusaslicer does not offer those.

### Affected users and/or printers

I believe many people would like this feature and I don't believe it would be hard to implement, especially since as I stated above, it was already implemented into a modified Cura.

### Additional information & file uploads

I want the z seam on this Skull model, for example, to be straight up and down. I cannot achieve this on the official version of Cura

[image]
.

smartavionics, 2022-07-30:
> A community member has released a modified version of Cura that has this feature. I love it. It works great. Unfortunately it is on Cura 4.10 which is now outdated.

More recent releases are available, now based on 4.13.1, see https://github.com/smartavionics/Cura/releases

tstone52, 2022-07-30:
> > A community member has released a modified version of Cura that has this feature. I love it. It works great. Unfortunately it is on Cura 4.10 which is now outdated.
> 
> More recent releases are available, now based on 4.13.1, see https://github.com/smartavionics/Cura/releases

First off, thank you for developing this! It has been quite helpful to many of my prints. How difficult and/or time consuming would it be to implement this into the latest Cura? I want it on at least Cura 5.0 in order to use the Arachne slicing engine. I have limited coding knowledge but might be willing to take a crack at it since I am trying to learn more about coding anyways.

smartavionics, 2022-07-30:
> How difficult and/or time consuming would it be to implement this into the latest Cura?

I haven't looked at the Arachne source so I can't say for sure but my guess is that it wouldn't be too hard as, conceptually, it's a simple enough modification.

Vandresc, 2022-08-01:
Hi @tstone52 , thank you for your feedback. We are indeed aware that the Z seam is not very optimal, we do have a ticket on our backlog, seems that will fix this sort of issue. Devs, CURA-9474 for our internal reference.

skumrask, 2023-01-09:
Any news on this, I want to calibrate couple of printers it's difficult to verify retraction and coasting when the seams are not aligned. I'll try with another shape now when I see this is a bug perhaps related to curved walls, but still a lot of time spent to figure out why my seams are so ugly, and completely fail when printing thinner walls. Doesn't seem to extrude enough after the seam when printing a thinner wall.

Sonicboom247, 2023-01-09:
agreed, i am trying to get the seam in the center of this piece since it will be hidden by another part.  It seems Cura does not care what is in the Z seam Position as the seam never moves.
[image]

pnallinger, 2023-02-21:
Sadly it is still looks like this. Wich looks horrible compared to a straight alinged seam...

[image]