Ultimaker/Cura #12900: Time Left

grumpyfart, 2022-07-30:
### Is your feature request related to a problem?

When I was using a Ender 5 Pro the display would show me how much time I had left on the Print
When I switched to the Ender 6 the display only shows me how much time has elapsed 

### Describe the solution you'd like

I think this would be a big help on the Ender 6 just like it was for me on the Ender 5 Pro

### Describe alternatives you've considered

the amount of time left for the print to finish

### Affected users and/or printers

anyone with a Ender 6 printer will benefit from this change

Asterchades, 2022-07-30:
There is a post processing script which can be used to display an estimated remaining time on a print. This will show the remaining time as a printer message, typically replacing the X/Y/Z coordinate display.

However I feel like the display you're familiar with was handled at the firmware level, not the slicer level. Specifically the SHOW_REMAINING_TIME flag in Configuration_adv, which was likely enabled on the Ender-5 but disabled on the Ender-6. Restoring that exact functionality will require compiling and uploading a custom-built firmware.

grumpyfart, 2022-07-31:
attached is the Plugin already in Cura
Itg works in the Ender 5 but doesn't work with the Ender 6

Asterchades, 2022-07-31:
Attachments don't seem to work if you simply reply to the email instead. I'm not sure if this is a recent thing or not but every time I've seen people try nothing is included, as is the case this time. Should you want to include an attachment it will need to be done directly via GitHub.

Just as I've said, however, the Ender-5 and Ender-6 run different firmwares. Due to the different kinematics it simply isn't possible to be otherwise. There's a flag in the firmware (SHOW_REMAINING_TIME) which is clearly set in the Ender-5 firmware that isn't set in the Ender-6 firmware (though there's no technical reason for this), and there's precisely nothing that can be done about that at the slicer level.

MariMakes, 2022-09-21:
Hey all, 

Thanks for the contribution GrumpyFart 😄 

I'll be closing this ticket. 
@Asterchades correctly indicates that this is a firmware issue and not one we can solve in Cura. 
I hope you understand.

closed 2022-09-21 by MariMakes