Ultimaker/Cura #12901: slicing fail

Ironkiki3000, 2022-07-30:
### Application Version

5.0.1

### Platform

windows 11

### Printer

cr10

### Reproduction steps

I tried slicing the deadpool helmet from do3d and every time it gives an error

### Actual results

[image]

### Expected results

it should slice properly 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Asterchades, 2022-07-30:
Without a project file the only assumption can be that the model is in need of repairs. Cura 5 appears to be more sensitive to broken models than its predecessors.

There were also some specific circumstances related to slicing failures that were addressed with the 5.1 update so it may even be worth updating and trying again.

Ironkiki3000, 2022-08-01:
hello I tried upgrading to 5.1 still it didnt work

Asterchades, 2022-08-01:
In that case a project file will be essential for any further diagnostics. A log might help as well.

Ironkiki3000, 2022-08-01:
Hello ,
How do I get a log of the error ?could you please help .

Ironkiki3000, 2022-08-01:
Is this what were yoy asking for ? please check the attachment

 
  

 
 

    
Listening for engine communications on 49674
Closing down
Listening for engine communications on 49674
Exception in thread Thread-18:
Traceback (most recent call last):
  File "C:\Users\farha\AppData\Roaming\Creality Slicer\Creality Slicer\python\lib\threading.py", line 552, in __bootstrap_inner
    self.run()
  File "C:\Users\farha\AppData\Roaming\Creality Slicer\Creality Slicer\python\lib\threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "Cura\util\sliceEngine.py", line 319, in _runEngine
    for k, v in self._engineSettings(extruderCount).iteritems():
  File "Cura\util\sliceEngine.py", line 539, in _engineSettings
    settings['fanFullOnLayerNr'] = (fanFullHeight - settings['initialLayerThickness'] - 1) / settings['layerThickness'] + 1
ZeroDivisionError: integer division or modulo by zero

Running: C:\Users\farha\AppData\Roaming\Creality Slicer\Creality Slicer\CuraEngine.exe -v -p -s initialSpeedupLayers=4 -s minimalFeedrate=10 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.
;This code is added before the T(n)
 -s supportXYDistance=700 -s insetXSpeed=35 -s retractionZHop=0 -s extruderOffset[3].X=0 -s extruderOffset[3].Y=0 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.
;This code is added after the T(n)
 -s retractionSpeed=80 -s filamentFlow=100 -s max_acceleration[1]=9000000 -s infillOverlap=15 -s skinSpeed=80 -s inset0Speed=35 -s coolHeadLift=0 -s extrusionWidth=400 -s upSkinCount=10000 -s initialLayerSpeed=20 -s max_xy_jerk=20000 -s minimalLayerTime=3 -s infillSpeed=80 -s supportExtruder=-1 -s fanSpeedMax=100 -s supportType=1 -s enableCombing=1 -s fanSpeedMin=50 -s supportZDistance=150 -s max_acceleration[3]=10000000 -s supportEverywhere=0 -s filamentDiameter=1750 -s initialLayerThickness=300 -s supportAngle=-1 -s fanFullOnLayerNr=0 -s acceleration=3000000 -s extruderOffset[1].X=0 -s extruderOffset[1].Y=0 -s layerThickness=320 -s endCode=;End GCode
M104 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)
G91                                    ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F4800 ;move Z up a bit and retract filament even more
G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
M84                         ;steppers off
G90                         ;absolute positioning
M81
;CURA_PROFILE_STRING:eNrtWltv2zYUfjWC/Qg+dljj6WI3aQ09rF3Sl3YokAxr8yLQEm1xkUSBpOJc4P++j6Qky6m9pWuwXiY/JNCnc8hz+c4liHN6w2ScMb7MdOSNw2C0onke64wnlyVTCtjxSDItaaK5KGNW0nnOonNZs5ESOU/j3J6wUfDHwWjBcUbKSsX1TeR73qiSvNSxqhhLo+P2UbOiYpLqWrIoCHagQbQDDHeBk13gtAPnLN267Zk3UnVVCamj30TJRlVO9ULIIqZpxhT8dHAjE6c1zWN2rWVt370UOhuteMViLVZMRqc0V6wHxFcirwsW+dORELcsVhlnedqIITS0YLAp5fitoe6Pj6Yfw8b3j8BwFzjZBU774CIXK5uGUtze5jCI3zIkdtJPbJebHkYLUZc6ejbeQm00mlf+s/G0/67gZYyHK5bDra03iSjmvFxGv+T5PQVebEXX274tE5WF5kJrUWwRMxxZ7nnxiqc6ixfQENL6KeZ/sgR84+WlVRZXTOa0sqYDQGKckRuvm+MdmR0KRvLSMtk9m4Ms5alkdKPJS8W01zyHU/d8vXlOhMhtXJpC4WBGOFrQtpbSppguOZiW85IhVDa0DbSkVRSaq+1TG6+clUudgWLGKnPYooahXR07zNkw7T3EBb22EeqsWgBFVYCtDZgxiqLmC90Q1lW5Rhp6Re/C5RAboibChvOoMm45iLJDUbHYRbFRb0pK31QsegNvVQfRcpn3SzO2Bzvbph14fQOOK03LxFD4qMNv+7CRr7ikuSF64wYvKlC/EGmLzGFmP+DIt6QLhJjKJS+j6bh5tiKqoomhb9iic6rYPTZucKNiSYkqaOTRhpgERbeVgqP7bzeqtkDNS8olSBCjL7vi2GDmhMABqm70DctUdA/ddWensXXjgl+j5qTkoGZcl7YNmIGAfMW0Tfh+kXnX5voyCImoWBnPuVa7BFD9ZlhcIc6a6yQzkXZiVV4jGcgQOLSM2sJOmIlXfB0d+vegG0A/gAdSj5eJyfXsLOcJSwnVL8hdSm/W5qdm+GUqcX0we0kVTwhqVuNW9YK8MQEkrpKgkvcG5Jr8AUchc7c9JNfkFFQF3J97OPqdGT7E3IN3zXhyl542rZkgNKlTdK3a9dV10YNW7u5lTysRSve1zDNOfes/98jZ3c6Ztyaz30s0YauvBaFpSm5ELYlYlQSypCdLDCXMcd7z7riHHvXRMa8DnzSfGdqCRLCvaF4zdfAa1rZv6ByNpdaMVALBA6WQjIO3x0EngAQROyRSZMdc2iqYij6AqUedpMk+WXGdEZ0xgt5HxGIBO47Je4988IwdaFfk/c8fzEFogoSVqUKHU1bowmuMNUIXH4v45MJHcyCnd/0hsm4UzJXtOkFSExF/WhTwNSAnnbubz+yWSWGVGudS4lq7uec08DxyEt7TaARJWBRwjJgGTmBacvnplxC6pLy0V207A+7WmtjMIxGkAMnpkhFRkjevfiUqkYyVCLp/RN41MuPxeIQYtVV3Uqbk9asmNRNytsOqni+m3lCU0ibqrT/x9ipYudQSdqNCnvCFYSDJ4ALh+kdEwicP+MwkQ6rQd7Zoh3CcHCIkoef9g7LdUhzL2tKkBE2OuMlHzBg18TMSbvd6aviEWxnGBFGiYCaFljMSMUbRWH795I2nsGFK3h8GoKz5sZttF6SumispIt5a1FnDrliJCjHHbuj/t07tK42nsNYaavYDwhURIEhj+4reoFYn+w9VmlXoCsoV4vP9NuzrA/7BDH1IwC8zIkC55Xpk6zwY2vwjtPnJ57d5Ny3O/V0nBf/qKO/Lzp6vfaKc7+lws7OVWaHMLUYzQFtou+znTCHf2z2FfO9Rx5C96/Bu95+b64Nz70FOL7hU+lty+5Gmb7Bn/J57nziBjY4/TO1han/G1N4zscNhYg8T+/85sYMHDa9QflcTe1hThjXl3poSfsk1xegEw2ozrDaPv9pMhtXmq1ptgsdbbYYt6T/bksIHzc4JjP6OFoZhNRxWw2E1LNPJt7YaGp1wWCeHdfJx18nmWy39bzJ04ObfrO4bO1tCFulJSIYZm7Bxoq6iEZLg2sxJw8tuYXXNZ870CpVpnU5qKW2IWwqbBNhEA+nQp2SVQaGrdLtMFHWueZV37UKq8cHsPENQzW0muFhvLMsti8yh50/KH0eIif6a7KMLU4CteX8Btoj2Qg==
 -s minimalExtrusionBeforeRetraction=0 -s retractionMinimalDistance=1500 -s skirtMinLength=150000 -s objectSink=0 -s retractionAmount=6000 -s nozzleSize=400 -s skirtLineCount=1 -s startCode=M190 S60.000000
M109 S220.000000
;Sliced at: Fri 18-02-2022 16:13:00
;Basic settings: Layer height: 0.32 Walls: 0.8 Fill: 100
;Print time: #P_TIME#
;Filament used: #F_AMNT#m #F_WGHT#g
;Filament cost: #F_COST#
;M190 S60 ;Uncomment to add your own bed temperature line
;M109 S220 ;Uncomment to add your own temperature line
G21        ;metric values
[... cut ...]
This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
alo

Vandresc, 2022-08-02:
Hi @Ironkiki3000, you can save a project file containing the current configuration which reproduces the error, from the File menu > Save a project. Then you can zip the file and attach it to this ticket. This will help us to reproduce the error you encounter. Thank you!

Ironkiki3000, 2022-08-04:
hello ,
hope this helps

LLK1_MAIN DEADPOOL TEXTURED from Do3D.zip

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
LLK1_MAIN DEADPOOL TEXTURED from Do3D.zip

Shared with Dropbox
 |

 |

 |

Vandresc, 2022-08-05:
Hi @Ironkiki3000, unfortunately, it doesn't work if you reply via email. You will need to access github in order to attach it.

Ironkiki3000, 2022-08-05:
Hello, 
Thank you for reaching back , alright i will upload it to the GitHub page

Ironkiki3000, 2022-10-11:
Hello the file is 85mb and i can't upload to github

Ironkiki3000, 2022-10-11:
Hello , 
I tried both 5.0 and 5.1 and everytime i am getting the same error . Can it be software issue ? When i try to print half of the file then it can slice but as soon as i try to slice the whole file it gives me the error .

Sincerely
Ironkiki3000

Asterchades, 2022-10-11:
Absolutely it could be a software issue. As I mentioned there were some failures which 5.1 corrected, and I believe 5.2 (currently in beta) corrects even more of them - and the inevitable 5.3 will doubtless correct even more, and so forth. It's a perpetually moving goal post, however, as more intricate and complicated models create more opportunities for both modelling issues and edge cases to interfere. 85mb is certainly larger than anything I've ever created, though most everything I've ever made isn't large enough to need that kind of fidelity.

If at all possible, could you try uploading the file to a 3rd party hosting service and provide a link? The service referenced in the initial template is WeTransfer, but anything which allows other people to download would be suitable (eg Dropbox, Google Drive, Mega, Mediafire, etc).

closed 2022-10-25 by github-actions[bot]