Ultimaker/Cura #12905: Cura 5.1 - Gradual Infill Steps parameter causing infill path defects on infill types that DONT use that parameter

botagar, 2022-07-30:
### Application Version

5.1

### Platform

Ubuntu 22.04

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Select `Cubic` as the infill pattern
2. Set the `Gradual Infill Steps` parameter to a non 0 value (ie. 5)
3. Select an infill pattern that does NOT use the `Gradual Infill Steps` (ie. `Cubic Subdivision` or `Cross 3D`)
4. Slice

### Actual results

Gcode generated is full of repeats and tiny little segments in very small areas resulting in an overflow of plastic being extruded on almost every layer.

### Expected results

Infill paths in Gcode dont repeat or have strange tiny line artefacts

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The initial narrowing down of the issue is outlined in this cura forum thread: https://community.ultimaker.com/topic/41138-cura-51-pathing-for-infill-jittery-constantly-backtracking-causes-catastrophic-print-issues/
[Project file and stl](https://github.com/Ultimaker/Cura/files/9226211/cura_issue.zip)
[Gif of issue](https://user-images.githubusercontent.com/1773846/181910106-e9e70619-74a6-488c-8f1f-d75a9be2b73a.gif)

GregValiant, 2022-07-30:
Thanks for the report.
I can confirm this is happening.  The screenshot below shows the number of short line segments being piled on top of one another.
[image]

jellespijker, 2022-08-09:
Devs see CURA-8926 already on our backlog