Ultimaker/Cura #12906: slicing error

WannaBeTinker, 2022-07-30:
### Application Version

5.1

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

slicing a purchased model from myminifactory.com
rotating the model with the "by face": and picking the bottom of the model

### Actual results

[TMPL_1S_Floor_D.zip](https://github.com/Ultimaker/Cura/files/9226841/TMPL_1S_Floor_D.zip)

### Expected results

failed slicing with error dialog
[AV_TMPL_1S_Floor_D.zip](https://github.com/Ultimaker/Cura/files/9226845/AV_TMPL_1S_Floor_D.zip)
g

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]

GregValiant, 2022-07-30:
Thanks for the report.
The model isn't "watertight" (meaning it has some open seams) and there are other errors.  It sliced fine for me as it is.  I used .2 layer height and .4 line width.
You should repair the model (MS 3D Builder has a decent repair capability) and then try again.
Regarding that pattern you show on the bottom, it looks like the bottom of the model isn't resting flat on the build plate.  That didn't show up when I opened it.
I'll leave this as a bug report for now so you can respond but I can't duplicate the problem and that model has definite issues.

WannaBeTinker, 2022-07-31:
Thanks for the quick response. go ahead and close the bug report. I just
havent run into this before and just upgraded to this version.   I also
tried taking the STL and doing a repair on it and opening the repaired file
and also got the same issues so not sure why my system was throwing them.

thanks
John

GregValiant, 2022-07-31:
I'm going to leave it open.  Maybe the Cura team will find something I missed.