Ultimaker/Cura #12907: 5.1.0 slicing failed.

joneyes, 2022-07-30:
### Application Version

5.1.0

### Platform

Win 10

### Printer

Ender 5 Pro

### Reproduction steps

slicing the attached file fails everytime
[CE5_Egg_Skelter_Mid.zip](https://github.com/Ultimaker/Cura/files/9227595/CE5_Egg_Skelter_Mid.zip)
Original STL and project file  included

### Actual results

Fails with slicing Failed
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/9227592/cura.log)

### Expected results

slicing should complete

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

.

joneyes, 2022-07-30:
Added DX Diag:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/9227598/DxDiag.txt)
Also tried importing into fusion 360, converting to solid and regenerating the stl - still failed.

GregValiant, 2022-07-30:
Thanks for the report.
When models fail to slice I always check them for errors.  In this case the Mesh Tools in Cura didn't find anything wrong and neither did MS 3D Builder.
When I uploaded to [Formware](https://formware.co/OnlineStlRepair) it also didn't report any errors but gave this result.
-> Vertex count changed from 6791 to 6681 (-110)
-> Triangle count changed from 13574 to 13354 (-220)
Since it made changes to the file I downloaded the "repaired" file and it sliced no problem.
I see this once in a while where a supposedly "good" model isn't so good.  Those cases are rare (no errors but there are errors) and this is one of them.
I will leave this report open.  I will also attach the file that sliced.  Let me know if it works for you.
[Egg_Skelter_Mid_fixed.zip](https://github.com/Ultimaker/Cura/files/9227611/Egg_Skelter_Mid_fixed.zip)
.

joneyes, 2022-07-31:
Many thanks Greg - thanks for the tips.
This seems to have fixed it.

closed 2022-07-31 by joneyes