Ultimaker/Cura #12908: 5.1.0 Not responding when trying to install profile update

Dean1782, 2022-07-30:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3s1 Pro

### Reproduction steps

Clicked on install update under profile name
Program then opens a blank marketplace screen and then does not respond

### Actual results

Program freezes and does not rrespond and needs to be shut down

### Expected results

Updates installed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No adfditional info

Vandresc, 2022-08-01:
Hi @Dean1782 , could you perhaps share your log file with us? And are you using an external monitor?

Dean1782, 2022-08-12:
Hi I can get it to work when i disconnect my monitors and just use the laptop. This is a pain as I dont want to have to disconnect m9onitors

Vandresc, 2022-08-15:
We're still investigating this, at present moment, we have two (other) leads, both largely outside of our sphere of direct influence.
Could you try the workaround from https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232 and see if that works out for you?

The other lead will be a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070
Which we reported and awaiting a fix from their side.

Duplicate of https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232