Ultimaker/Cura #12910: Unexpected Slicing Bug

I522I, 2022-07-31:
### Application Version

5.1.0

### Platform

Win 10, R5 5600X, RX590, 16GB

### Printer

Artillery Genius

### Reproduction steps

1. Load model
2. start slicing

### Actual results

Unexpected Error; no slicing; I should report a bug, the error said.

### Expected results

succesfully slicing.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9228640/cura.log)
[AG_toothless.zip](https://github.com/Ultimaker/Cura/files/9228641/AG_toothless.zip)

GregValiant, 2022-07-31:
Thanks for the report.
The 3mf file you included is not a project file so it's hard to tell what is going on without having you settings.  Use the "File | Save Project" command and that type of 3mf file will include your printer, the model, and your settings.

I522I, 2022-08-01:
Oh, sorry, dont know, why its not a project, but now it is. I checked.
Slicing still not working, so a system restart is not the solution. ;)
[AG_toothless.zip](https://github.com/Ultimaker/Cura/files/9235774/AG_toothless.zip)

GregValiant, 2022-08-01:
Thanks for that.
That model consists of 21 separate meshes and although it is water-tight there are errors.
I opened the file in MS 3D Builder and it noted errors in the model and repaired them.  It still would not slice.
I then merged the 21 separate pieces in 3D Builder and the model finally sliced.
So this is a problem with the model.  The way it was put together without being merged was odd.
I'll leave this open for a bit so you can respond but I don't see a bug here.

I522I, 2022-08-03:
Ok, well, thank you.
I didn't see that as a reason, cause I have the mesh analysing plugin (don't know the exactly name, I'm not at home) installed and the Plugin didn't show any problems.
But I don't know if the plugin is from Ultimaker or third party.
So thanks for Ur fast analyse of my files!

GregValiant, 2022-08-03:
In the Mesh Tools section of the Cura settings, if you turn off "Union overlapping Volumes" and turn on "Remove all Holes" it will slice, but the slice is weird.  The toes don't get connected and there is something strange about the eyes as well.  This model was not assembled correctly.  The best bet is to merge all the separate pieces using another piece of software as I described above.
This is the best I could do just with Cura.  This would fall apart as the separate pieces don't go together well.
[image]

GregValiant, 2026-08-19:
I'll close this as a model problem.

closed 2026-08-19 by GregValiant