Ultimaker/Cura #12913: Gyroid infill shows past the cutting mesh (gyroid pot model)

CrazySensei, 2022-07-31:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

-Imported 2 models -one pot body named "gyroid top" and one more identical pot body with pattern holes on the side walls named "top".
-I centered both models on the bed.
- Set "gyroid top" with "per model settings" to cutting mesh.
- Set "top" with the ff "per model settings":  walls = 0, top and bottom layers = 0, infill pattern = gyroid, infill distance = 1mm
- Slice

I do these exact same steps using the exact same model with Cura 4.13(and previous models as well). The result should be a pot with gyroid patterns on its side walls.

The problem:
In Cura 5.1.0, it generates the same result BUT with the gyroid infills sticking out from the cutting mesh parts ever so slightly. 
It makes the model look like it had short hair growing all over.

I have tried reinstalling Cura about four times and made sure that the model in fusion is on point -it is.
So, I uninstalled 5.1.0 and installed 4.13 to recheck everything. I believe it is a bug. 

### Actual results

In Cura 5.1.0, it generates the same result BUT with the gyroid infills sticking out from the cutting mesh parts ever so slightly. 
It makes the model look like it had short hair growing all over.

### Expected results

I do these exact same steps using the exact same model with Cura 4.13(and previous models as well). The result should be a pot with gyroid patterns on its side walls.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-07-31:
Thanks for the report.
We need a project file in order to take a look at this.  Load your models, set Cura up to slice, use the "File | Save Project" command to create a 3mf file.  Zip the 3mf file and post the zip file here.

CrazySensei, 2022-07-31:
Thank you for the quick response..Sorry I dont have winzip buthere is the 3mf file..

GregValiant, 2022-07-31:
You need to post the 3mf file here.  Emails don't work with GitHub as files don't show up.
If you right-click on the file and select the "Send To" option one of the sub-menus might be "Compressed (Zip) Folder".  That will work.

CrazySensei, 2022-08-01:
[CE3MAX_Top.zip](https://github.com/Ultimaker/Cura/files/9232564/CE3MAX_Top.zip)

Got it.
Thanks for the assist.
Here it is..

CrazySensei, 2022-08-01:
Got it..
Here is the file..
[CE3MAX_Top.zip](https://github.com/Ultimaker/Cura/files/9232582/CE3MAX_Top.zip)

GregValiant, 2022-08-01:
Thank you.
What you are seeing is Expected Behavior rather than a bug.
Your "gyroid" model is X=90.8 x Y=90.8 and is larger than the "top" model which is 88.9 x 88.9.  That results in the "gyroid" model printing beyond the outside surfaces of "top".  (Maybe you got the scaling backwards?)
You can add "Horizontal Expansion" to the cutting mesh that is "top" and then set it to 0.5 to get this to print the way you expect.
Let us know if that works.  I'll leave the bug label for now but this appears to be a model/settings issue.

closed 2022-08-15 by github-actions[bot]