Ultimaker/Cura #12914: Prepare and preview show different content.

nlkirchn, 2022-07-31:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality CR10

### Reproduction steps

Wanted to print the Labyrinth Gift box. Loaded the Lid file and sliced, changed to Preview and noticed that it didn't show the same profile of the object. 

I reinstalled the 4.1 version of Cura and went through the same steps and the Preview version and the Prepare version ae the same and the sliced file was correct and printed as such.

[image]
[image]
 

### Actual results

When I saved the slice it printed the Preview version instead of the Prepare version.

### Expected results

It should print the prepared version.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I can't send a project or original stl file as it saves with a 3mf type or stl which this app does not support.
[cura.log](https://github.com/Ultimaker/Cura/files/9229125/cura.log)

GregValiant, 2022-07-31:
Thanks for the report.
A project file will contain the model, your settings, and your printer.  We really need that to investigate what is going on.
With the model loaded and Cura set up ready to slice select the "File | Save Project" command to create the 3mf file.  Then select "Show in Folder".  Right-click on the 3mf file and select "Send To" and then "Compressed (Zip) Folder".  Post the zip folder here.

nlkirchn, 2022-07-31:
I hope this is what you need.

From: GregValiant
Sent: Sunday, July 31, 2022 4:14 PM
To: Ultimaker/Cura
Cc: nlkirchn; Author
Subject: Re: [Ultimaker/Cura] Prepare and preview show different content.(Issue #12914)

Thanks for the report.
A project file will contain the model, your settings, and your printer. We really need that to investigate what is going on.
With the model loaded and Cura set up ready to slice select the "File | Save Project" command to create the 3mf file. Then select "Show in Folder". Right-click on the 3mf file and select "Send To" and then "Compressed (Zip) Folder". Post the zip folder here.

GregValiant, 2022-07-31:
GitHub doesn't accept attachments to emails.  They don't make it through.
You need to post the file(s) here directly on the website.

nlkirchn, 2022-07-31:
Try this.

From: GregValiant
Sent: Sunday, July 31, 2022 5:57 PM
To: Ultimaker/Cura
Cc: nlkirchn; Author
Subject: Re: [Ultimaker/Cura] Prepare and preview show different content.(Issue #12914)

GitHub doesn't accept attachments to emails. They don't make it through.
You need to post the file(s) here directly on the website.

nlkirchn, 2022-07-31:
I guess I don’t understand how to get you the file. Send me an email from Utilimaker so I can attach a file.

From: GregValiant
Sent: Sunday, July 31, 2022 5:57 PM
To: Ultimaker/Cura
Cc: nlkirchn; Author
Subject: Re: [Ultimaker/Cura] Prepare and preview show different content.(Issue #12914)

GitHub doesn't accept attachments to emails. They don't make it through.
You need to post the file(s) here directly on the website.

nlkirchn, 2022-07-31:
[CCR10_labyrinthGiftBox_lid.zip](https://github.com/Ultimaker/Cura/files/9229869/CCR10_labyrinthGiftBox_lid.zip)
[CCR10_labyrinthGiftBox_lid.zip](https://github.com/Ultimaker/Cura/files/9229872/CCR10_labyrinthGiftBox_lid.zip)

nlkirchn, 2022-07-31:
Maybe this works.

GregValiant, 2022-07-31:
Under the Mesh Fixes - turn off "Remove all Holes" because it's like...removing all holes.
I'll close this as it's a settings issue.  (I should have seen it right off but the zoom was insufficient.)

closed 2022-07-31 by GregValiant