Ultimaker/Cura #12915: Cura Desktop Icon does not have Image on Gnome for YEARS.

leoheck, 2022-07-31:
### Application Version

5.1.0

### Platform

Ubuntu 22.04

### Printer

-

### Reproduction steps

-

### Actual results

-

### Expected results

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This still happens on 5.1.0. 
Come on folks, fix this. It is not difficult. 
And can you remove "Ultimaker" in the Name of the icon in the .desktop file?
It never shows the version that is hidden. So I have no idea which application this is if I have more than one versions, which is always a good idea when upgrading to a new version, since thing may break.

[image]

jellespijker, 2022-07-31:
How is it installed?
I suspect you used a system package manager since the AppImage has an icon. We can only support the AppImage, due to limited resources. If it is installed with an system package manager, please contact the maintainers of that package.

leoheck, 2022-07-31:
It was always installed using the packages provide here.

leoheck, 2022-07-31:
What is the recommended way to install the package? 

I am assuming that making it executable and double-clicking it or launching it from the terminal is the right way of installing it.

Other packages work out of the box. They install with an icon that comes with the appimage

MariMakes, 2022-08-26:
Hey @leoheck,

I don't really know how to help you. 🤔 
Most Linux folks I know download and install the AppImage from the release page.
https://github.com/Ultimaker/Cura/releases/tag/5.1.0

I'm not really familiar with the issue you are describing. 
I asked our Linux developers and they do not recognize your problem.
That's probably why it's never gotten priority to fix. 

If it's not difficult to fix you are welcome to make a pull request yourself.
We are an open-source project after all. 😊

leoheck, 2022-08-26:
> Most Linux folks I know download and install the AppImage from the release page.

This is what I do.

> I asked our Linux developers and they do not recognize your problem.

Which system do they use? 
I use the standard Ubuntu 2022.04 and not any other unknown alternative. 

> If it's not difficult to fix you are welcome to make a pull request yourself.

It is not difficult for me to investigate and propose a fix, for sure. 
Thanks for replying and for letting me know, that the user help is appreciated.

jellespijker, 2022-08-26:
I personally use Cura on my Ubuntu 22.04 KDE and Gnome, a Manjaro KDE, and an Arch Linux with Xfce.
All without issues, the other devs use Ubuntu and Manjaro I believe.

Keep in mind that the AppImage isn't an installer, it is a stand alone application. Unlike our Windows and Mac distributions the AppImage will not install itself on your system. It will just start up Cura.

I personally download the AppImage, give it executable rights and make sure it is located on a directory on my `PATH`
I manually add the AppImage to my menu items.

The resources which are used to build the AppImage can be found here: https://github.com/Ultimaker/Cura/tree/main/packaging/AppImage

And the icons here: https://github.com/Ultimaker/Cura/tree/main/packaging/icons

leoheck, 2022-08-27:
Please, try any other app image. And you will see how it works.

darrellenns, 2022-12-31:
This seems to be an issue affecting many AppImage packages. If there is only an icon in the root of the AppImage and not under /usr/share/icons, then the icon is not used properly when integrating with the desktop. I believe the simple fix would be to just have the AppImage include freedesktop style icons in /usr/share/icons. PrusaSlicer and SuperSlicer are affected by the same issue.

See also:
AppImageCommunity/libappimage#165
TheAssassin/AppImageLauncher#475
prusa3d/PrusaSlicer#7801
supermerill/SuperSlicer#2594

jellespijker, 2023-01-01:
Thanks @darrellens for doing some research on this. I will add this to my todo list, if you feel adventurous you might want to open up a PR for this fix. I think you would only need to change "something" here: https://github.com/Ultimaker/Cura/blob/75c0533ef8faf8884ea5896e277b1bac6127a885/packaging/AppImage/create_appimage.py#L28

commit e8ef90da (2023-04-28): Merge pull request #15015 from leoheck/fix-appimage-icons

Fix AppImage icons on some Linux systems

Fixes #12915 
Fixes #14662

Implements CURA-10550
closed 2023-04-28 by jellespijker