Ultimaker/Cura #12917: slicer failed after saving

msteelman, 2022-07-31:
### Application Version

5.1.0

### Platform

Mac OS

### Printer

Geeetech A20

### Reproduction steps

I rotated the two objects to -84 degrees and then saved. Then it said something had changed and asked me if I wanted to reload. When I reloaded the objects they changed. There had been two different objects and now it was two copies of the same object. I closed the file and reopened it and the models appeared to be right. I tried to slice and the slicer failed and recommended I report the bug.

### Actual results

slicer failed

### Expected results

file would slice as it had at least twice previously

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_FDM_100_top.3mf.zip](https://github.com/Ultimaker/Cura/files/9229420/CFFFP_FDM_100_top.3mf.zip)

GregValiant, 2022-07-31:
Thanks for the report.
Everything runs in streaks.  This is the second time I've come across this problem in the last couple of days.
There are no errors in the model and yet there are errors in the model.  I used [Formware.co](https://formware.co/OnlineStlRepair) to check the model and this is the report from there:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 59796ms ------
-> Vertex count changed from 151515 to 159540 (+8025)
-> Triangle count changed from 304322 to 326158 (+21836)
You can see that although it didn't find any of the common errors it still increased the triangle and vertex counts.  Something was off in there.

Using your models I was able to export each model from Cura as STL's and bringing them in separately they sliced fine.  This reminds me of an issue in the beta version that I think was worked on.  The orientation of the models either rotationally or by translating to somewhere different on the build plate affected Cura's ability to slice the model.  The Cura team will take a look.

msteelman, 2022-07-31:
If I rewind my workflow another step, I had used the function align to build plate soon before I had the problem. When I try to use that function it snaps the mesh onto the other mesh. Not sure if that is intended. I didn’t think a lot of it at the time, I changed the view to the top and moved it back where I wanted it.

Regards,
Mark

GregValiant, 2022-07-31:
I don't think so.  The orientation of a model will produce different toolpaths but I wouldn't think it should affect Cura's ability to slice the model.  That's what was happening in the Beta.  Maybe something needs another tweak.

MariMakes, 2022-08-26:
Hey @msteelman,

Welcome to Github 🎉 

I recognize your issue 🔍 
It seems to be a duplicate of: https://github.com/Ultimaker/Cura/issues/12089
It's related to overwriting the files you already have on your build plate.
We have a ticket on the backlog with the intent to make the behavior clearer.

I'm closing this issue since it's a duplicate, you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12089

closed 2022-08-26 by MariMakes