Ultimaker/Cura #12918: GET /api/v1/system HTTP/1.1\r\n fails returns HTTP/1.1 404 Not Found\r\n

Mclute0, 2022-07-31:
### Application Version

5.1.0

### Platform

windows 10

### Printer

MP MINI DELTA V2

### Reproduction steps

1) boot up printer connected to WiFi that is verified to work with Wiibuilder 2.1.6.0 that uses CuraEngine
2) open cura and attempt to connect "add network printer by IP" network printer
3) connection fails

### Actual results

connection failes.

wireshark shows 

0000   e0 e2 e6 9c 57 c0 50 eb f6 24 d4 bb 08 00 45 00   ....W.P..$....E.
0010   00 e9 52 5e 40 00 80 06 00 00 c0 a8 c8 f3 c0 a8   ..R^@...........
0020   c8 e5 c9 17 00 50 29 4f d4 97 00 0d b4 3f 50 18   .....P)O.....?P.
0030   fc 20 14 06 00 00 47 45 54 20 2f 61 70 69 2f 76   . ....GET /api/v
0040   31 2f 73 79 73 74 65 6d 20 48 54 54 50 2f 31 2e   1/system HTTP/1.
0050   31 0d 0a 48 6f 73 74 3a 20 31 39 32 2e 31 36 38   1..Host: 192.168
0060   2e 32 30 30 2e 32 32 39 0d 0a 43 6f 6e 74 65 6e   .200.229..Conten
0070   74 2d 54 79 70 65 3a 20 61 70 70 6c 69 63 61 74   t-Type: applicat
0080   69 6f 6e 2f 6a 73 6f 6e 0d 0a 43 6f 6e 6e 65 63   ion/json..Connec
0090   74 69 6f 6e 3a 20 4b 65 65 70 2d 41 6c 69 76 65   tion: Keep-Alive
00a0   0d 0a 41 63 63 65 70 74 2d 45 6e 63 6f 64 69 6e   ..Accept-Encodin
00b0   67 3a 20 67 7a 69 70 2c 20 64 65 66 6c 61 74 65   g: gzip, deflate
00c0   0d 0a 41 63 63 65 70 74 2d 4c 61 6e 67 75 61 67   ..Accept-Languag
00d0   65 3a 20 65 6e 2d 55 53 2c 2a 0d 0a 55 73 65 72   e: en-US,*..User
00e0   2d 41 67 65 6e 74 3a 20 4d 6f 7a 69 6c 6c 61 2f   -Agent: Mozilla/
00f0   35 2e 30 0d 0a 0d 0a                              5.0....

Frame 50: 77 bytes on wire (616 bits), 77 bytes captured (616 bits) on interface \Device\NPF_{1CC02BAE-552F-4487-9095-1EFB89D167DF}, id 0
Ethernet II, Src: Espressi_9c:57:c0 (e0:e2:e6:9c:57:c0), Dst: ASUSTekC_24:d4:bb (50:eb:f6:24:d4:bb)
Internet Protocol Version 4, Src: 192.168.200.229, Dst: 192.168.200.243
Transmission Control Protocol, Src Port: 80, Dst Port: 51479, Seq: 72, Ack: 194, Len: 23
[3 Reassembled TCP Segments (94 bytes): #47(69), #48(2), #50(23)]
Hypertext Transfer Protocol
    HTTP/1.1 404 Not Found\r\n
        [Expert Info (Chat/Sequence): HTTP/1.1 404 Not Found\r\n]
            [HTTP/1.1 404 Not Found\r\n]
            [Severity level: Chat]
            [Group: Sequence]
        Response Version: HTTP/1.1
        Status Code: 404
        [Status Code Description: Not Found]
        Response Phrase: Not Found
    Content-Type: text/html\r\n
    Content-Length: 23\r\n
    \r\n
    [HTTP response 1/1]
    [Time since request: 0.008835000 seconds]
    [Request in frame: 46]
    [Request URI: http://192.168.200.229/api/v1/system]
    File Data: 23 bytes
Line-based text data: text/html (1 lines)
    This URI does not exist

### Expected results

this action works on diffently cura engine backed app...

0000   e0 e2 e6 9c 57 c0 50 eb f6 24 d4 bb 08 00 45 00   ....W.P..$....E.
0010   00 d2 4e f2 40 00 80 06 00 00 c0 a8 c8 f3 c0 a8   ..N.@...........
0020   c8 e5 c8 f3 00 50 ea ce 48 ab 00 0b 91 5b 50 18   .....P..H....[P.
0030   fc 20 13 ef 00 00 47 45 54 20 2f 61 70 69 2f 70   . ....GET /api/p
0040   72 69 6e 74 65 72 20 48 54 54 50 2f 31 2e 31 0d   rinter HTTP/1.1.
0050   0a 41 63 63 65 70 74 3a 20 61 70 70 6c 69 63 61   .Accept: applica
0060   74 69 6f 6e 2f 6a 73 6f 6e 0d 0a 41 63 63 65 70   tion/json..Accep
0070   74 2d 45 6e 63 6f 64 69 6e 67 3a 20 67 7a 69 70   t-Encoding: gzip
0080   0d 0a 58 2d 41 70 69 2d 4b 65 79 3a 20 30 39 37   ..X-Api-Key: 097
0090   45 42 45 44 41 31 33 41 31 34 30 32 42 42 32 44   EBEDA13A1402BB2D
00a0   30 37 35 30 35 38 37 34 45 36 34 31 34 0d 0a 48   07505874E6414..H
00b0   6f 73 74 3a 20 31 39 32 2e 31 36 38 2e 32 30 30   ost: 192.168.200
00c0   2e 32 32 39 0d 0a 43 6f 6e 6e 65 63 74 69 6f 6e   .229..Connection
00d0   3a 20 4b 65 65 70 2d 41 6c 69 76 65 0d 0a 0d 0a   : Keep-Alive....

0000   50 eb f6 24 d4 bb e0 e2 e6 9c 57 c0 08 00 45 00   P..$......W...E.
0010   01 5d 05 0e 00 00 ff 06 a2 62 c0 a8 c8 e5 c0 a8   .].......b......
0020   c8 f3 00 50 d7 20 00 01 db 01 ef 89 22 3e 50 18   ...P. ......">P.
0030   15 c6 75 8b 00 00 0d 0a 7b 22 73 64 22 3a 7b 22   ..u.....{"sd":{"
0040   72 65 61 64 79 22 3a 74 72 75 65 7d 2c 22 73 74   ready":true},"st
0050   61 74 65 22 3a 7b 22 66 6c 61 67 73 22 3a 7b 22   ate":{"flags":{"
0060   63 61 6e 63 65 6c 6c 69 6e 67 22 3a 66 61 6c 73   cancelling":fals
0070   65 2c 22 63 6c 6f 73 65 64 4f 72 45 72 72 6f 72   e,"closedOrError
0080   22 3a 66 61 6c 73 65 2c 22 65 72 72 6f 72 22 3a   ":false,"error":
0090   66 61 6c 73 65 2c 22 66 69 6e 69 73 68 69 6e 67   false,"finishing
00a0   22 3a 66 61 6c 73 65 2c 22 6f 70 65 72 61 74 69   ":false,"operati
00b0   6f 6e 61 6c 22 3a 74 72 75 65 2c 22 70 61 75 73   onal":true,"paus
00c0   65 64 22 3a 66 61 6c 73 65 2c 22 70 61 75 73 69   ed":false,"pausi
00d0   6e 67 22 3a 66 61 6c 73 65 2c 22 70 72 69 6e 74   ng":false,"print
00e0   69 6e 67 22 3a 66 61 6c 73 65 2c 22 72 65 61 64   ing":false,"read
00f0   79 22 3a 74 72 75 65 2c 22 72 65 73 75 6d 69 6e   y":true,"resumin
0100   67 22 3a 66 61 6c 73 65 2c 22 73 64 52 65 61 64   g":false,"sdRead
0110   79 22 3a 74 72 75 65 7d 2c 22 74 65 78 74 22 3a   y":true},"text":
0120   22 4f 70 65 72 61 74 69 6f 6e 61 6c 22 7d 2c 22   "Operational"},"
0130   74 65 6d 70 65 72 61 74 75 72 65 22 3a 7b 22 74   temperature":{"t
0140   6f 6f 6c 30 22 3a 7b 22 61 63 74 75 61 6c 22 3a   ool0":{"actual":
0150   32 35 2c 22 6f 66 66 73 65 74 22 3a 30 2c 22 74   25,"offset":0,"t
0160   61 72 67 65 74 22 3a 30 7d 7d 7d                  arget":0}}}

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

this wifi connection is via an expressif esp32 wroom32e
bluetooth is not enabled in the esp32 firmware
connecting via usb to com7 port works as expected from the wiibuilder app described but attempting connecting via cura 5.1.0 to USB com port on CH340a chip results in continuous reboots. don't think that issue is related

Is there an easy edit to json configs to resolve this query error: I am not sure why one way works but the other returns E404. I need to dig into the Wiibuilder code to see why theirs is working with...

cura engine works using 

GET /api/printer HTTP/1.1\r\n 

cura 5.1.0 fails using 

GET /api/v1/system HTTP/1.1\r\n

[image]

Mclute0, 2022-08-21:
so I physically cut the wire from ch340g DTR pin to MCU nRST pin and now after only being able to print from WiiBuilder, I can now print directly from Cura 5.1.0 via USB. One works so web api doesn't really matter. very bad printer design with DTR to nRST, maybe was intended to push the manufacturers lesser quality slicer.

jellespijker, 2022-08-23:
Sorry for the delayed response.
Out-of-the-box network support is only for Ultimaker Printers. The used API of your third-party printer differs too much from ours. Connecting your printer will only work if the company or one of the community members writes a plugin for Cura which adds network support.

I hope you understand that I will close this issue because Ultimaker itself can't use resources into adding native network support for third-party printers themself.

closed 2022-08-23 by jellespijker