Ultimaker/Cura #12919: Application Crashes when opening Marketplace

JesNoFear, 2022-08-01:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality CR-10S Pro v2

### Reproduction steps

1. Load Application
2. Click x's on missing plugins (Just upgraded)
3. Click on Marketplace
4. window pops up but blank, trying to click out of it and the app crashes

### Actual results

Application Crashes

### Expected results

Marketplace window should open with content so i can download my plugins

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

application works on my dell precision laptop, but not on my optiplex desktop (7010)
side note: Desktop has 6 monitors, grid format, 3 on top 3 on bottom.

JesNoFear, 2022-08-01:
[cura.log](https://github.com/Ultimaker/Cura/files/9231071/cura.log)

Vandresc, 2022-08-01:
Hi @JesNoFear, were you connected to an external monitor when the issue occurred? And if so, does it still becomes unresponsive/crash when not using one? Thank you!

JesNoFear, 2022-08-03:
No, I have 2 video cards that support upto 4 monitors each.
AMD Radeon Pro WX4100 and AMD Radeon Pro WX3200

Vandresc, 2022-08-05:
We're still investigating this, at present moment, we have two (other) leads, both largely outside of our sphere of direct influence. 
Could you try the workaround from https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232 and see if that works out for you? 

The other lead will be a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070
Which we reported and awaiting a fix from their side.

Vandresc, 2022-08-05:
Probably a duplicate of https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232

JesNoFear, 2022-08-05:
So, looking at my services, i do not have the listed service on my computer. Attached Screenshot.

[image]

I am attaching a SYSINFO export, to help support your case with Qt, from what i am skimming, because i have 6 of the same monitor model is causing the issue.

[SYSINFO.txt](https://github.com/Ultimaker/Cura/files/9271423/SYSINFO.txt)

[image]

Let me know if you need anymore info or would like me to try something else.

JesNoFear, 2022-08-06:
So, some more details.
[image]

I was noticing, the splash screen was always showing up on monitor 6, main app would open on screen 2, market place and cura backup would open on screen 6.

So, in thinking it was having an issue seeking which monitor the dialog boxes are opening up with, i moved the main app to monitor 6, closed and re-opened and now i get a new error.

[image]

[stderr.log](https://github.com/Ultimaker/Cura/files/9275648/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/9275649/cura.log)
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/9275655/cura.cfg.txt) (removed hostname and keys_cache under octoprint)

JesNoFear, 2022-08-06:
Also, boyfriends computer has 4 monitors, using a single AMD Radeon Pro WX3200 video card, he has 4 monitors are same vendor, two are same model, and the other two are the same model as well. (Viewsonic)
Cura 5.1.0 runs fine on his computer even between monitors.

JesNoFear, 2022-08-06:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/9275735/DxDiag.txt)

JesNoFear, 2022-08-29:
Any update on this?

Vandresc, 2022-08-30:
@JesNoFear, unfortunately not :( Follow https://github.com/Ultimaker/Cura/issues/12264#issuecomment-1231227277 for more info.

GregValiant, 2026-08-19:
Is this still a problem in Cura 5.13.0??

JesNoFear, 2026-08-19:
No clue, software was not fixed in a reasonable amount of time, and migrated to superslicer and have never gone back.

GregValiant, 2026-08-19:
Then I'll close this.

closed 2026-08-19 by GregValiant