Ultimaker/Cura #12920: Allow changing multiple objects with "Per Model Settings"

gjsmo, 2022-08-01:
### Is your feature request related to a problem?

When printing many different objects, many objects may need custom settings, different from the rest. For example, only 10 of 100 parts need supports, but all can fit on the build plate at once. Currently, if multiple objects are selected, Per Model Settings can not be adjusted at all. This makes the task repetitive and annoying, despite all changes being the same.

### Describe the solution you'd like

Allow selection of multiple objects, then Per Model Settings. A fancy "conflict resolution" function isn't needed IMHO, if selecting multiple parts have conflicting settings then overwrite them to all be the same. In particular, there should be the ability to ADD settings as well as SET them.

### Describe alternatives you've considered

Parts must be individually selected, then settings are added, then they are set, and repeat. Very long and unnecessary.

### Affected users and/or printers

Primarily users with many small different parts, e.g. miniatures, customized merchandise, printing parts for Vorons.

Vandresc, 2022-08-02:
Hi @gjsmo , thank you for your feedback. I will bring your request with the rest of the team.

MariMakes, 2022-09-21:
Hey @gjsmo, 

Thank you for your suggestion 😄
We added a ticket on our backlog.
Fingers crossed 🤞 
For internal reference: CURA-9529