Ultimaker/Cura #12922: SLICING ERROR

brenozicarelli, 2022-08-01:
### Application Version

5.1.0

### Platform

win 10

### Printer

Kingroon KP3s

### Reproduction steps

1. load model
2. start slicing
3. error

### Actual results

error return code 3221226505

### Expected results

successful slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

22-08-01 03:03:51,620 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [88]: Adding QT6 plugin path: C:\Program Files\Ultimaker Cura 5.1.0\PyQt6\plugins
2022-08-01 03:03:51,620 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [98]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 5.1.0\PyQt6\plugins
2022-08-01 03:03:51,620 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-08-01 03:03:51,620 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 5.1.0
2022-08-01 03:03:51,620 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 8.1.0
2022-08-01 03:03:51,621 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2022-08-01 03:03:51,621 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib\uranium must be a folder.
2022-08-01 03:03:51,621 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib64\uranium must be a folder.
2022-08-01 03:03:51,621 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib32\uranium must be a folder.
2022-08-01 03:03:51,621 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Ultimaker Cura 5.1.0\plugins must be a folder.
2022-08-01 03:03:51,621 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Resources\uranium\plugins must be a folder.
2022-08-01 03:03:51,621 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Resources\cura\plugins must be a folder.
2022-08-01 03:03:51,622 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
2022-08-01 03:03:51,622 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is C:\Users\breno\AppData\Roaming\cura\5.1
2022-08-01 03:03:51,622 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is C:\Users\breno\AppData\Roaming\cura\5.1
2022-08-01 03:03:51,622 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is C:\Users\breno\AppData\Local\cura\5.1\cache
2022-08-01 03:03:51,622 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.1.0\share\uranium\resources\bundled_packages\uranium.json
2022-08-01 03:03:51,622 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\bundled_packages\cura.json
2022-08-01 03:03:51,622 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2022-08-01 03:03:51,622 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2022-08-01 03:03:51,623 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [191]: Detected most suitable OpenGL context version: 4.1 Core profile
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.Qt.QtApplication.initialize [198]: Initializing job queue ...
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.Qt.QtApplication.initialize [202]: Initializing version upgrade manager ...
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 5.1.0\share\uranium\resources\bundled_packages\uranium.json
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 5.1.0\share\cura\resources\bundled_packages\cura.json
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.PackageManager._loadManagementData [201]: Loaded user packages management file from C:\Users\breno\AppData\Roaming\cura\5.1\packages.json
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 
2022-08-01 03:03:51,623 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [121]: Loading plugin configuration file 'C:\Users\breno\AppData\Roaming\cura\5.1\plugins.json'
2022-08-01 03:03:51,624 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib\cura must be a folder.
2022-08-01 03:03:51,624 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin FileLogger 1.0.1
2022-08-01 03:03:51,640 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin LocalContainerProvider 1.0.1
2022-08-01 03:03:51,659 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin LocalFileOutputDevice 1.0.1
2022-08-01 03:03:51,674 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin CameraTool 1.0.1
2022-08-01 03:03:51,691 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin MirrorTool 1.0.1
2022-08-01 03:03:51,712 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin RotateTool 1.0.1
2022-08-01 03:03:51,731 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin ScaleTool 1.0.1
2022-08-01 03:03:51,744 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin SelectionTool 1.0.1
[... 93 lines cut ...]
2022-08-01 03:03:54,122 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin OrientationPlugin 3.7.1
2022-08-01 03:03:54,123 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [416]: Loading all plugins took 2.63189959526062 seconds
2022-08-01 03:03:54,135 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2022-08-01 03:03:54,185 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [177]: Checking and performing updates took 0.054049015045166016
2022-08-01 03:03:54,194 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [265]: Completed loading preferences.
2022-08-01 03:03:54,231 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [300]: Created system tray icon.
2022-08-01 03:03:58,082 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [548]: Loading metadata into container registry took 3.8335092067718506 seconds
2022-08-01 03:03:58,094 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0
2022-08-01 03:03:58,101 - INFO - [MainThread] cura.CuraApplication.run [823]: Initializing machine error checker
2022-08-01 03:03:58,307 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [308]: The definition cache for definition kingroon_kp3s failed to pickle.
2022-08-01 03:03:58,314 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [kingroon_kp3s]
2022-08-01 03:03:58,463 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [308]: The definition cache for definition kingroon_base failed to pickle.
2022-08-01 03:03:58,512 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [192]: Extruder[kingroon_base_extruder_0 #2] added to [KP3S] at position [0]
2022-08-01 03:03:58,513 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name Kingroon KP3S in list of firmware to check.
2022-08-01 03:03:58,726 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for kingroon_kp3s took 0.20747637748718262 seconds.
2022-08-01 03:03:58,742 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: ChangeAtZ
2022-08-01 03:03:58,747 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: ColorMix
2022-08-01 03:03:58,751 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: CreateThumbnail
[... cut ...]
2022-08-01 03:03:58,768 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: InsertAtLayerChange
2022-08-01 03:03:58,774 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: PauseAtHeight
2022-08-01 03:03:58,778 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: RetractContinue
2022-08-01 03:03:58,782 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: SearchAndReplace

Vandresc, 2022-08-01:
Hi @brenozicarelli , could you please share a project file so we can reproduce the issue described by you? Thank you! 

You can save it as a project with the configuration that reproduces the issue from the File menu, zip the file and attach it to this ticket, please.

brenozicarelli, 2022-08-01:
[files.zip](https://github.com/Ultimaker/Cura/files/9235622/files.zip)

Vandresc, 2022-08-02:
For me, it successfully slices when loading the project you attached. Also, the log doesn't provide any information regarding the slicing error, this makes it hard to address if we cannot reproduce it.

brenozicarelli, 2022-08-02:
sorry for getting in the way, the last windows update somehow caused several programs to stop working, I reversed the upgrade and CURA is working normally.

jellespijker, 2022-08-02:
Great to hear that it works again. I will close this ticket

closed 2022-08-02 by jellespijker