Ultimaker/Cura #12923: Support renders other Layers invisible in preview

Waldkaeuzchen, 2022-08-01:
### Application Version

5.10

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

Slice any modell with support enabled

### Actual results

In the preview the modell becomes translucent behind the support-structure
[image]

### Expected results

I would expect the modell to not be translucent or to see the walls of the modell behind the support if it should be translucent. This doesnt effect the atual G-Code, it seems to just be an optical Bug.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[invisible_layers.zip](https://github.com/Ultimaker/Cura/files/9233223/invisible_layers.zip)

GregValiant, 2022-08-01:
Thanks for the report.
As you get the preview closer to exactly straight on to the layers then this can happen.  The cyan colored support is in front of the red shell and the Z for each layer is exactly the same.  There is no "Send to back" capability but you can turn off the "Helper" linetype to view the shell.
Can you explain some more about your computer?  Information regarding the chip set and graphics card would be useful.
This is the view I get at about 45°.
[image]

And this is straight on in Orthographic view.  I don't' see a bug here as this is what I would expect to see considering the infill is in front of the shell in that area.
[image]

Waldkaeuzchen, 2022-08-02:
Hi,

here are some details of the system:
CPU: I5-8265U
GPU: Intel UHD Grafics 620
Chipset:  9D84 for Intel(R) 300 Series Chipset Family 

Here is another image with a steeper angle, again you can see the Text from the buildplate (the "M" from the Ultimaker-Buildplate in this case), so the Layers a rendered different than on your machine.
[image]

Vandresc, 2022-08-02:
Hi all, this is somehow expected behavior, though I can understand the confusion. This occurs when the model is in front of the backplate, then the model becomes translucent as the focus is the support.