Ultimaker/Cura #12928: Custom wall ordering

fkiliver, 2022-08-02:
### Is your feature request related to a problem?

Now Cura's choice of wall order is only from inside to outside and from outside to inside.However, in some cases, taking the 3-layer wall as an example, better surface quality can be obtained by using the order of inside-outside-middle.
Like this
|    |    |
|    |    |
|    |    |
2  3  1   (wall ordering)
outside-middle-inside

### Describe the solution you'd like

In the wall ordering option, add a custom printing order for each wall or add a ordering that two sides first and then the middle.
|    |    |
|    |    |
|    |    |
3  2   1   (wall ordering)
outside-middle-inside

### Describe alternatives you've considered

Add an extra lead line inside the model from the outside to the inside.

### Affected users and/or printers

Anyone

Vandresc, 2022-08-05:
Hi @fkiliver, to me your feature request looks similar to the behavior described in https://github.com/Ultimaker/Cura/issues/11934.  If that is the case then the "center line" green line will always be printed last as it is not the inner wall, so Outside to Inside and Inside to Outside may look indeed confusing.

MariMakes, 2022-09-21:
We changed the description so hopefully it's clearer what to expect from the wall ordering. 🤞 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/11934

closed 2022-09-21 by MariMakes